mantidproject/mantid #38464: Improve validation when applying calibration in ISISIndirectEnergyReduction algorithm

adriazalvarez, 2024-12-03:
Reported by: Sanghamitra 

**Describe the bug**
ISIS Indirect Energy Reduction Algorithm can apply a calibration step by dividing the spectra to a correction factor contained on a calibration file. 
If the calibration file has some values equal to 0, it results on a divide by zero error when the calibration is applied and the final
reduction is not valid. 

**To Reproduce**
This script should run the algorithm just fine: 
```
test_calib=LoadEmptyInstrument(InstrumentName="IRIS")
test_calib=CropWorkspace(test_calib,StartWorkspaceIndex=2, EndWorkspaceIndex=52)
#test_calib.setY(0,np.array([0]))

ISISIndirectEnergyTransfer(InputFiles='IRS26173.raw', Instrument='IRIS', Analyser='graphite', 
                           Reflection='002', Efixed=1.845, SpectraRange='3,53', 
                           FoldMultipleFrames=False, GroupingMethod='All',
                           CalibrationWorkspace='test_calib',OutputWorkspace='out_red')
UnGroupWorkspace(InputWorkspace='out_red')
```
Now uncommenting the line : `test_calib.setY(0,np.array([0]))` should run the algorithm wihout errors, but the output reduction file `irs26173_graphite002_red` does not contain any value. 

**Expected behavior**

User is notified that calibration file contains invalid values, those values are replaced by closer to 0 values when possible. 
See how is handled in ISISIndirectDiffraction algorithm for vanadium containers.
**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

SpencerHowells, 2024-12-04:
@adriazalvarez This needs further investigation. If there is a zero that detector/spectrum should not be used.

robertapplin, 2024-12-04:
Spencer found that the calibration file has faulty detectors at 50 and 51.

A user can exclude faulty detectors manually using the "Custom" detector grouping option. In this case we can use the "Custom" detector grouping, and select "3:49,52:53"

robertapplin, 2024-12-04:
Thanks for writing this up! I think this issue highlights the need for more validation or a warning message when using a calibration with faulty detectors, and I am happy for that to be considered a bug fix

I believe the ability for faulty detectors to be identified and excluded from a data reduction should be considered a New Feature. We wanted to work on this feature earlier this year, however there were communication difficulties. See this issue: https://github.com/mantidproject/mantid/issues/36377

PR #38663 (2025-01-27): Add warning for zero-valued calibration spectra
### Description of work

If a calibration workspace is provided to `ISISIndirectEnergyTransfer` with any spectra that are zero, then the algorithm will now output a warning listing the workspace indices with zeroes.

#### Summary of work

Added warning, release note, test.

Fixes #38464 . 

### To test:
See #38464 for an example of how to test this, check the logger output for a warning if you've got a zero.
closed 2025-01-27 by github-actions[bot]