mantidproject/mantid #38597: Documentation build fails on non-linux machines

MialLewis, 2025-01-14:
A documentation build currently fails on non-linux machines:

```
6>CUSTOMBUILD : warning : autodoc: failed to import class 'AlgoTimeRegisterImpl' from module 'mantid.api'; the following exception was raised:
6>Traceback (most recent call last):
6>  File "C:\Users\wym92549\AppData\Local\mambaforge\envs\mantid-developer\lib\site-packages\sphinx\util\inspect.py", line 397, in safe_getattr
6>    return getattr(obj, name, *defargs)
6>AttributeError: module 'mantid.api' has no attribute 'AlgoTimeRegisterImpl'
6>
6>The above exception was the direct cause of the following exception:
6>
6>Traceback (most recent call last):
6>  File "C:\Users\wym92549\AppData\Local\mambaforge\envs\mantid-developer\lib\site-packages\sphinx\ext\autodoc\importer.py", line 207, in import_object
6>    obj = attrgetter(obj, mangled_name)
6>  File "C:\Users\wym92549\AppData\Local\mambaforge\envs\mantid-developer\lib\site-packages\sphinx\ext\autodoc\__init__.py", line 343, in get_attr
6>    return autodoc_attrgetter(self.env.app, obj, name, *defargs)
6>  File "C:\Users\wym92549\AppData\Local\mambaforge\envs\mantid-developer\lib\site-packages\sphinx\ext\autodoc\__init__.py", line 2835, in autodoc_attrgetter
6>    return safe_getattr(obj, name, *defargs)
6>  File "C:\Users\wym92549\AppData\Local\mambaforge\envs\mantid-developer\lib\site-packages\sphinx\util\inspect.py", line 413, in safe_getattr
6>    raise AttributeError(name) from exc
6>AttributeError: AlgoTimeRegisterImpl
```

It looks to be that `sphinx` `autodoc` is trying to import `AlgoTimeRegisterImpl`, which is only made available to the python api on Linux:

https://github.com/mantidproject/mantid/pull/38175/files#diff-3fa2df896a03358080fa6e2320c6cbb4060e495c003b054577b4e1240e82b409R93

MialLewis, 2025-01-14:
@mpatrou is there any reason why `AlgoTimeRegisterImpl` can't be made available to all OS'?

sf1919, 2025-01-14:
@jclarkeSTFC is looking into this I believe so I have assigned him

PR #38601 (2025-01-16): Include AlgoTimeRegister on all operating systems
Include `AlgoTimeRegister` on all operating systems. It was originally only included on Linux, see #38175.

By only including `AlgoTimeRegister` on Linux, we need to either edit the docs to only have examples that only work on Linux, or selectively include the doc page for that class. It doesn't do any harm to use `AlgoTimeRegister` on other operating systems, then the tests can be run on all OS's, and things are a bit neater.

Fixes #38597 

Things that needed fixing to get everything working on Windows and macOS:

- On Windows x64, a `long` i [... cut]
closed 2025-01-16 by github-actions[bot]