mantidproject/mantid #38598: Unable to load data if a TimeSeriesProperty contains only a single time

rbauststfc, 2025-01-14:
**Original reporter:** Max Skoda and John Webster from the ISIS Reflectometry Group

**Describe the bug**

Attempting to load run `INTER81632` fails with error "Cannot guess ending value from a TimeSeriesProperty that contains only a single time".

**To Reproduce**

Run the following:

```
ws = Load("INTER81632")
```

**Expected behavior**

Prior to version 6.10 of Mantid, the workspace would still have been loaded with all the sample logs. This can be seen by running the above in version 6.9 of Mantid. It's unclear why this behaviour was changed (if it was intentional, or is a regression), but our Reflectometry scientists feel that the log should still be loaded in the way it used to be.

**Platform/Version (please complete the following information):**
 - Mantid Version: this seems to be a regression introduced in Mantid version 6.10

**Additional context**

The issue with this run appears to be in the `Nima_mode` SE log. If you look at entry `selog->Nima_mode->value_log->time` in the raw Nexus file (load into an HDF viewer) then there is only one positive time entry. This seems to be different to run `INTER81633`, which loads OK and has two positive time entries. Further investigation of the code would be needed to confirm that this is definitely the part of this entry that is causing the problem, but it seems consistent with the error message (and the location in the code that is throwing that error).

I am confident that the `Nima_mode` log is the cause of the problem as running the below script (which excludes this log) works OK:

```
ws = LoadEventNexus(Filename="INTER81632", LoadMonitors=True, BlockList="Nima_mode")
```

It wasn't particularly easy to track this down from the error message, as it doesn't mention which Nexus file entry was being loaded at the time of the problem.

John has explained that he sees this occasionally whenever he uses a NIMA trough in a Reflectometry experiment (at a frequency of approximately one run in three experiments).

rbauststfc, 2025-01-15:
Max has been doing some comparisons of different Nexus file log entries and this seems to show that the number of positive time entries is not what is determining how many time series entries get loaded. I've now looked further into the code and I think (although this still needs further investigation to confirm) it may be to do with the number of valid entries. If you look at entry `selog->Nima_mode->value_log->value_valid` in the raw Nexus file then you can see that the `Nima_mode` log for `INTER81632` has only one value marked as valid. The log loading code looks like it includes a step where it filters out invalid values (in method `PropertyManager::filterByProperty`), which would in theory leave us with only a single time series value for the `Nima_mode` log for `INTER81632`. Further investigation is required to confirm that this is really why we are getting the load error for this run.

rbauststfc, 2025-01-20:
FYI @MialLewis this one came up last week from Reflectometry, but Tom thinks the Core Team will have time to look at it so it's on their to-do list at the moment.

mducle, 2025-03-14:
I found a MERLIN datafile which also cannot be loaded due to this bug unless you do:

```
www = LoadEventNexus('MER70731.nxs', BlockList='triton_closed_loop,triton_d,triton_heater_range')
```

The file is on Babylon: `\\Olympic\Babylon5\Public\duc\MER70731.nxs` (43MB)

PR #39870 (2025-09-15): Catch runtime errors when adding period logs
### Description of work

Inconsistencies sometimes occur in period-related logs within Event NeXus files, such as duplicated entries with the same timestamps. Therefore, these repeated entries are reduced to a single entry. This can lead to runtime errors when loading the file because it is not possible to determine the ending value from a TimeSeriesProperty with a single time entry. In the past, this has resulted in Mantid crashes. In order to prevent this, the related event workspace is now created without the period logs if issues are detected.

 [... cut]
closed 2025-09-15 by jclarkeSTFC