mantidproject/mantid #38629: Manual Testing Indirect interfaces

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:
Follow the directions at: 

* https://developer.mantidproject.org/Testing/Indirect/DiffractionTests.html
* https://developer.mantidproject.org/Testing/Indirect/DataReductionTests.html

jclarkeSTFC, 2025-01-21:
Can't get the archive to work on my mac so can't test this on macOS

GuiMacielPereira, 2025-01-21:
No issues on Linux, took ~10 mins

jclarkeSTFC, 2025-01-21:
No problem on Windows, about 15 minutes

jhaigh0, 2025-01-22:
No problem on IDAAAS ~15 mins (It is annoying to get the file from unit test data though, could it just be added to the sample data?)

closed 2025-01-22 by jclarkeSTFC