mantidproject/mantid #38630: Manual Testing Inelastic interfaces

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

"Follow the directions at:

* https://developer.mantidproject.org/Testing/Inelastic/BayesFittingTests.html
* https://developer.mantidproject.org/Testing/Inelastic/CorrectionsTests.html
* https://developer.mantidproject.org/Testing/Inelastic/QENSFittingTests.html 
* https://developer.mantidproject.org/Testing/Inelastic/DataProcessorTests.html
"

MialLewis, 2025-01-21:
- BayesFittingTests [ ] 
- CorrectionsTests [ ]
- QENSFittingTests [ ]
- DataProcessorTests [ ]

MialLewis, 2025-01-21:
QENSFittingTests, s(q,w). Saw this error when closing sliceviewer. Seems inconsistent, occurs when cursor is over the plot:
'''
Traceback (most recent call last):  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/matplotlib/cbook.py", line 298, in process    func(*args, **kwargs)  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/widgets/sliceviewer/views/cursor.py", line 57, in on_mouse_leave    self.on_cursor_outside_axes()  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/widgets/sliceviewer/presenters/imageinfowidget.py", line 67, in on_cursor_outside_axes    self._widget.cursorAt(DBLMAX, DBLMAX, DBLMAX, extra_cols)RuntimeError: wrapped C/C++ object of type ImageInfoWidget has been deleted
'''

MialLewis, 2025-01-21:
Data processor test s(q,w). ILL data `243489-243506ElwinHeat_H2O-HSA-stw.nxs` is in system test data, inconvenient.

MialLewis, 2025-01-22:
Issues raised.

closed 2025-01-22 by MialLewis