mantidproject/mantid #38632: Manual Testing Workbench Core

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

jclarkeSTFC, 2025-01-21:
Not sure if this is a bug, but it took me a while to figure out why the Superplot interface was showing a blank graph for every spectrum, and it was because it opened zoomed in, so the trick was to click the home button to see the plot.

GuiMacielPereira, 2025-01-21:
Took around ~45 mins, I tried to be quite thorough.

jclarkeSTFC, 2025-01-21:
Windows no problems, took around 45 mins

closed 2025-01-21 by jclarkeSTFC