mantidproject/mantid #38633: Manual Testing Muon

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

See instructions at: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html

RichardWaiteSTFC, 2025-01-21:
Testing took ~ 2 hours

Found a couple of issues in the Muon Analysis UI that are also present in v6.11 - will check for regressions and update this comment

- Struggling to find data on archive
- On plot in LHS can specify XMin > XMax 

I also saw an issue with NaN errors on IDAaaS in the simultaneous fit (of Normal type after having done a TF Aysm. fit) in [MUSR] test (https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_MUSR.html#transverse-field-asymmetry-test) but this was not very reproducible (also wasn't seen on other OS) so I haven't opened an issue.

Also tempted to open an issue for clarifying the testing instructions: the user needs to press enter to load a run, the load current run doesn't load the run-number entered in the box.

RichardWaiteSTFC, 2025-01-21:
@sf1919 issue #38356 may not be a regression but if there is an issue and the user can't load data then Mantid is useless!

sf1919, 2025-01-22:
@RichardWaiteSTFC do you mean #38656 ? The one you reference is a closed issue for the data profiling Epic.

SilkeSchomann, 2025-01-22:
The potential Qt issue turned out to be not causing the delays when loading data. Also, it is not a regression from Mantid 6.11. Therefore, I don't think it is worthwhile raising an issue for it.