mantidproject/mantid #38634: Manual Testing TOFConverter

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

"Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
https://docs.mantidproject.org/interfaces/utility/TOF%20Converter.html "

RichardWaiteSTFC, 2025-01-21:
Time to test: 10 mins

Found a few issues (not regressions - present in v6.11):
- Float division by 0 error when two-theta=0 error message could be clearer (like the existing check for input value > 0)
- Large number sometimes produces `Numerical result out of range` error, and other larger numbers just return `inf` with no error

Also reproduced (closed/stale) issues
- #36082
- #36083

RichardWaiteSTFC, 2025-01-21:
Closing as issues raised

closed 2025-01-21 by RichardWaiteSTFC