mantidproject/mantid #38639: Manual Testing ISIS SANS

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html

jclarkeSTFC, 2025-01-21:
Can't run the reduction on macOS, I'm getting errors that look like this:
```
Error in execution of algorithm SANSReductionCore:
Invalid value for property Filename (string) from string "DIRECT.041": When setting value of property "Filename": File "DIRECT.041" not found
 at line 116 in '/Applications/MantidWorkbenchNightly.app/Contents/Resources/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSReductionCore.py'
```
Might be archive problems, but I thought the stuff in the training data folder was self-contained.

jhaigh0, 2025-01-21:
~1hour

GuiMacielPereira, 2025-01-21:
All working, Linux, ~1h15m

jclarkeSTFC, 2025-01-22:
Windows no problems, maybe about an hour

closed 2025-01-22 by jclarkeSTFC