mantidproject/mantid #38641: Manual Testing Mantid Basics Course

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

* Check the Qt-help docs, built into Workbench, against the online documentation 
* Open up the basic course: http://docs.mantidproject.org/tutorials/mantid_basic_course/index.html#mantid-basic-course
* Check that the information and exercises are up to date and work in the Workbench release candidate

rbauststfc, 2025-01-21:
[This screenshot](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/02_displaying_and_navigating.html#settings) of the Display Settings drop-down seems to be missing some options so could be updated.

rbauststfc, 2025-01-21:
[This screenshot](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/04_masking_and_grouping.html#masking) in the Draw tab of the instrument view needs updating.

rbauststfc, 2025-01-21:
Get the following unexpected exception on the Filter Events Interface:

> Traceback (most recent call last):  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqtinterfaces/FilterEvents/eventFilterGUI.py", line 216, in callback    sender_edit.callback()  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqtinterfaces/FilterEvents/eventFilterGUI.py", line 286, in set_startTime    newtime0 = np.clip(inps, xlim[0], float(self.ui.leStopTime.text()))
> ValueError: could not convert string to float: ''

Reproduce by:

- Go to `Interfaces -> Utility -> Filter Events`.
- Delete the entry in the `Stopping Time` field (under the plot). Leave the entry in the `Starting Time` field.
- Put your cursor into the `Starting Time` field and hit enter on the keyboard.

Also works the other way around (i.e. clear `Starting Time` and hit enter in `Stopping Time`). The dialog seemed to change a bit in 6.10 and the issue was introduced then. Note that the screenshot [here](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/06_sample_logs.html#interface) is also showing the old version of the interface and needs to be updated.

rbauststfc, 2025-01-22:
Some of the screenshots on [this page](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#save-settings) need updating.

rbauststfc, 2025-01-22:
I've opened issue #38664 to capture updates to the course documentation.

MohamedAlmaki, 2025-01-22:
The loading screenshot is missing clear [here](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/01_loading_data.html#loading-data)

MialLewis, 2025-01-22:
[Displaying data](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/02_displaying_and_navigating.html#displaying-and-navigating) page suggests using `MUSR00015189` dataset.

When using the pick tab as per [Investigating data](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html#investigating-data), the unit of the x axis is `label` on the miniplot.

This causes an error to be thrown when using the `Add a single crystal peak` tool, which may confuse new users.

Is this wierd unit being caused by a regression?

MialLewis, 2025-01-22:
When selecting multiple detectors using a drawn shape, are we expecting to see any details in the `selection` box?

rbauststfc, 2025-01-22:
> When selecting multiple detectors using a drawn shape, are we expecting to see any details in the `selection` box?

If I try doing this in 6.9 then there don't seem to be any details either.

MialLewis, 2025-01-22:
This is now finished from my perspective.

closed 2025-01-23 by rbauststfc