mantidproject/mantid #38642: Manual Testing M-slice

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Direct/MSliceTestGuide.html

cailafinn, 2025-01-21:
@SilkeSchomann We're having some issues with M-Slice versions on Windows & macOS. 

Windows and macOS (Both Standalone): `2.10.1+uncommitted` (`nightly` & `main` label)
Linux (Conda & Standalone): `2.10.1.dev25+uncommitted` (`nightly` label)

Windows and macOS seem to be targeting the main label, rather than the nightly.

sf1919, 2025-01-21:
There is an issue already open for this so any details you can add to #38628 would be welcome

SilkeSchomann, 2025-01-21:
> [@SilkeSchomann](https://github.com/SilkeSchomann) We're having some issues with M-Slice versions on Windows & macOS.
>
> Windows and macOS (Both Standalone): `2.10.1+uncommitted` (`nightly` & `main` [...]

For a Conda installation, you could just install the MSlice nightly in your Conda environment. For non-Conda installations the only option is to search for the `mslice` folder for your installation and replace it with https://github.com/mantidproject/mslice/tree/main/src/mslice

thomashampson, 2025-01-22:
There was a mistake in the earlier report: it looks like standalone packages on all OS have the same version of mslice, and the conda installs on all OS have the latest nightly one. I have a full understanding of why and have commented on the issue.

cailafinn, 2025-01-22:
This dropdown to the right of `Bose` is crushed on macOS:

[image]

robertapplin, 2025-01-22:
The output plot looks slightly different for this test https://developer.mantidproject.org/Testing/Direct/MSliceTestGuide.html#changing-the-intensity-of-a-cut

The data has no margin at the low y and high y:

[image]

cailafinn, 2025-01-22:
UnEx on `Plot Over` [step](https://developer.mantidproject.org/Testing/Direct/MSliceTestGuide.html#modifying-a-cut): 

[image]

warunawickramasingha, 2025-01-22:
[4. Interactive Cuts](https://developer.mantidproject.org/Testing/Direct/MSliceTestGuide.html#id17) step 9, the instructions need to be updated, since it is the x axis  label that is changing when Click on Flip Integration Axis.

cailafinn, 2025-01-22:
All issues created. No regressions.

closed 2025-01-22 by cailafinn