mantidproject/mantid #38643: Manual Testing Project Recovery

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

jclarkeSTFC, 2025-01-21:
When project recovery attempts to load a plot, an error appears saying `A plot was unable to be loaded from the save file. Error: Line2D.set() got an unexpected keyword argument 'argType'`

To reproduce, run the script from test 1, plot some spectra, run `SegFault`, restart Workbench and click `Yes` in the project recovery window.

jclarkeSTFC, 2025-01-21:
macOS took 35 minutes

Despiix, 2025-01-21:
When I try to run this line `testing_directory=`C:\Users\rcj23313\Desktop\testing_directory` I get this error

`SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 2-3: truncated \UXXXXXXXX escape`

Not sure if it's my mistake or not

jclarkeSTFC, 2025-01-21:
> When I try to run this line `testing_directory="C:\Users\rcj23313\Desktop\testing_directory I get this error"`
> 
> `SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 2-3: truncated \UXXXXXXXX escape`
> 
> Not sure if it's my mistake or not

That string needs to be r"C:\Users ..." or you need double backslashes

jhaigh0, 2025-01-21:
> When project recovery attempts to load a plot, an error appears saying `A plot was unable to be loaded from the save file. Error: Line2D.set() got an unexpected keyword argument 'argType'`
> 
> To reproduce, run the script from test 1, plot some spectra, run `SegFault`, restart Workbench and click `Yes` in the project recovery window.
> 
> Possibly due to [#38486](https://github.com/mantidproject/mantid/pull/38486)

Same for me

GuiMacielPereira, 2025-01-21:
Got the same error as @jclarkeSTFC on Linux. Also at the very last test (number 8), the history of the workspace _did not_ just contain `Load`, the entire workspace history was present. Maybe someone else could try to reproduce?

Test took 35-40 mins

Despiix, 2025-01-21:
**Windows**

Test 4: It does reopen workspaces, plots and interfaces. However, the plots are both blank.  

[image]

Test 5: The first instance of mantid recovered the previous save and did not update to instance 1.

Time: ~50 min

jhaigh0, 2025-01-21:
~50 mins

closed 2025-01-21 by jclarkeSTFC