mantidproject/mantid #38645: Manual Testing Sliceviewer

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html

rbauststfc, 2025-01-21:
"When you zoom in to a region comprising only of bins with 0 counts it will set the color axis limits to (0.1,1) and force the scale to be linear"

This doesn't appear to be happening (when tested with event workspace)

rbauststfc, 2025-01-21:
Transposing the axes removes the cursor position (i.e. "Transpose the axes (as in step 6) and confirm the cursor position displayed is still correct (note if Track Cursor is checked you will need to click on the colorfill plot again" doesn't work).

rbauststfc, 2025-01-21:
Peak Overlay instruction 3 is incorrect. It should read:

It should change the slicepoint along the integrated momentum axis and zoom into the peak - e.g. in (X,Y) = (K,L) then the slicepoint along H will be set to 1 and there will be a cross at (0,0).

rbauststfc, 2025-01-21:
We couldn't find the "axis label" referred to in CutViewer Tool instruction 27: "Produce a non-axis aligned cut where the peak is not in the center of the x-range. Take the x-value at the peak maximum and plug it into the axis label - it should produce the HKL of the peak (0,0,1)."

rbauststfc, 2025-01-21:
Issue opened for updating testing instructions: https://github.com/mantidproject/mantid/issues/38649

closed 2025-01-21 by rbauststfc