mantidproject/mantid #38647: Manual Testing ALF

warunawickramasingha, 2025-01-21:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Direct/ALFViewTests.html

jhaigh0, 2025-01-21:
Completed with no problems, ~15mins

Despiix, 2025-01-21:
I cannot get `82301` to load. Therefore, I cannot proceed with the manual testing. (Windows)

GuiMacielPereira, 2025-01-21:
In Calculating Rotation Angle, I got `0.756302` instead of `0.877078` for the peak centre. Not sure if this counts as `approximately`?

jhaigh0, 2025-01-21:
> In Calculating Rotation Angle, I got `0.756302` instead of `0.877078` for the peak centre. Not sure if this counts as `approximately`?

For me the rotation angle was the same as in the docs (at least to 5 s.f). Maybe you had the wrong tubes selected?

jclarkeSTFC, 2025-01-21:
Unable to get the file to load on macOS so I had to stop testing

jclarkeSTFC, 2025-01-21:
On Windows, I got a peak centre of `0.711871` and a rotation angle of `1.02938`

jclarkeSTFC, 2025-01-21:
Windows about 15 minutes