mantidproject/mantid #38648: Manual Testing Issue Creation script is not working

warunawickramasingha, 2025-01-21:
**Describe the outcome that is desired.**

There is a script to automate manual testing issue creation at:
https://github.com/mantidproject/documents/tree/main/Project-Management/Tools/RoadmapUpdate

However the script found not working when creating manual test issues for mantid release 6.12 and raise errors as below

[image]

**Describe any solutions you are considering**

**Additional context**

sf1919, 2025-01-21:
I don't know if it's relevant but Artifact actions v3 were deprecated towards the end of 2024 see https://github.com/actions/upload-artifact/blob/main/docs/MIGRATION.md for details of migrating from v3 to v4 in case it is relevant.

MohamedAlmaki, 2025-01-23:
I did thorough testing using various GitHub tokens, including Waruna's, and everything worked well on my end. Waruna also tested his token in an IDaaS instance, and it worked properly there as well. Therefore, we conclude that the issue occurs only on Waruna's local PC, indicating that the script is working well.

closed 2025-01-23 by MohamedAlmaki