mantidproject/mantid #38649: Update sliceviewer manual testing instructions

rbauststfc, 2025-01-21:
- [ ] This instruction is slightly unclear: "7.c.Transpose the axes (as in step 6) and confirm the cursor position displayed is still correct (note if Track Cursor is checked you will need to click on the colorfill plot again)". It would help if we clarified that it means that after transposing then the cursor position should still be picked up correctly.

- [ ] Peak Overlay instruction 3 is incorrect. It should read:

> "It should change the slicepoint along the integrated momentum axis and zoom into the peak - e.g. in (X,Y) = (K,L) then the slicepoint along H will be set to 1 and there will be a cross at (0,0)."

- [ ] We couldn't find the "axis label" referred to in CutViewer Tool instruction 27: "Produce a non-axis aligned cut where the peak is not in the center of the x-range. Take the x-value at the peak maximum and plug it into the axis label - it should produce the HKL of the peak (0,0,1).". This instruction could do with being a bit clearer.

rbauststfc, 2025-01-24:
Thanks for taking a look @warunawickramasingha. As it sounds like you found from your testing, when we went through this we found that we couldn't get the result described in the instructions (i.e. cross at (0,0)) unless we used (X,Y) = (K,L) rather than (X,Y) = (H,K). I don't recall why at this point, but we concluded during testing that this was an issue with the test instructions rather than with the slice viewer.

I'm very much not an expert in the slice viewer, so it might be an idea to ask Richard to take a quick look to confirm. If he confirms that it is definitely just a test instructions problem, then I would guess that you could either update the wording to be what we have suggested, or you could update it to describe what you observed.

warunawickramasingha, 2025-01-24:
> I observed that the cross is located at (H, K) = (1, 0) in both the md_3D and md_4D workspaces. The value displayed on the L slider (assuming this refers to the slice point) is either 0 or 1, depending on the row selected in the peaks table. For example, if row 1 is clicked, L becomes 1.0, and if row 2 is clicked, L becomes 0.0. Can you shed some light whether that is what is expected to reflect in the test instructions?

Thanks for the input @rbauststfc!

@RichardWaiteSTFC could you please confirm whether the test instructions need an update with what I have observed here?
https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#ads-observer-for-peak-overlay:~:text=1%2C0%2C1)%20and%20(1%2C0%2C0).-,Double%20click%20a%20row,-It%20should%20change

Thanks!

RichardWaiteSTFC, 2025-01-24:
> Thanks for the input [@rbauststfc](https://github.com/rbauststfc)!
>
> [@RichardWaiteSTFC](https://github.com/RichardWaiteSTFC) could you please confirm whether the test instructions need an update [...]

So the peaks are at (H,K,L) = (1,0,1) and (1,0,0) - so if viewing the (X,Y) = (H, K) plane you will see a cross at (1,0) for slicepoint L=0 and 1. So I think it is indeed an error in the test instructions here it should be (1,0) not (0,0)
https://github.com/mantidproject/mantid/blob/516709ee15e24a168b6f8bfdb1b8c4ffac099512/dev-docs/source/Testing/SliceViewer/SliceViewer.rst?plain=1#L191-L193

warunawickramasingha, 2025-01-27:
Thanks @RichardWaiteSTFC. 

In addition, I am also not clear with how to follow [this test instruction ](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#check-non-orthogonal-view-is-disabled-for-non-q-axes:~:text=Produce%20a%20non%2Daxis%20aligned%20cut%20where%20the%20peak%20is%20not%20in%20the%20center%20of%20the%20x%2Drange.%20Take%20the%20x%2Dvalue%20at%20the%20peak%20maximum%20and%20plug%20it%20into%20the%20axis%20label%20%2D%20it%20should%20produce%20the%20HKL%20of%20the%20peak%20(0%2C0%2C1).
 )as Rachel as highlighted. 

> * We couldn't find the "axis label" referred to in CutViewer Tool instruction 27: "Produce a non-axis aligned cut where the peak is not in the center of the x-range. Take the x-value at the peak [...]

Could you please suggest an alternative wording there to clarify what exactly is to be done there?

Thanks!

RichardWaiteSTFC, 2025-01-27:
> Thanks [@RichardWaiteSTFC](https://github.com/RichardWaiteSTFC).
> 
> In addition, I am also not clear with how to follow [this test instruction ](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#check-non-orthogonal-view-is-disabled-for-non-q-axes:~:text=Produce%20a%20non%2Daxis%20aligned%20cut%20where%20the%20peak%20is%20not%20in%20the%20center%20of%20the%20x%2Drange.%20Take%20the%20x%2Dvalue%20at%20the%20peak%20maximum%20and%20plug%20it%20into%20the%20axis%20label%20%2D%20it%20should%20produce%20the%20HKL%20of%20the%20peak%20(0%2C0%2C1).)as Rachel has highlighted.
> 
> > * We couldn't find the "axis label" referred to in CutViewer Tool instruction 27: "Produce a non-axis aligned cut where the peak is not in the center of the x-range. Take the x-value at the peak maximum and plug it into the axis label - it should produce the HKL of the peak (0,0,1).". This instruction could do with being a bit clearer.
> 
> Could you please suggest an alternative wording there to clarify what exactly is to be done there?
> 
> Thanks!

Perhaps `x-axis label on the 1D plot?
Here is a screenshot to demonstrate

[image]

The peak in the 1D plot is at `x~0.25`
Thex-axis label of the 1D plot is `(0.0, -0.24 + 0.97x, 0.93 + 0.25x)` so if we plug in the value of `x` at the peak centre` we get `(0.0, -0.24 + 0.97*0.25, 0.93 + 0.25*0.25) = (0.0, 0.0025, 0.9925` which is very close to the actual peak HKL `(0,0,1)` - note it won't be exact because the numbers in the x-axis label are rounded.

warunawickramasingha, 2025-01-27:
Thanks a lot @RichardWaiteSTFC  for the clarifications. I have include all these changes in [this PR](https://github.com/mantidproject/mantid/pull/38694)

closed 2025-02-13 by cailafinn