mantidproject/mantid #38650: Plots not loading in project recovery

jclarkeSTFC, 2025-01-21:
Found during manual testing for 6.12

**Describe the bug**

When project recovery tries to open plots it fails with the error:
```
A plot was unable to be loaded from the save file. Error: Line2D.set() got an unexpected keyword argument 'argType'
```

**To Reproduce**
Use the longer script from the first test [here](http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html), then plot a spectrum. Once project recovery has saved, crash Workbench with `SegFault`, restart Workbench, then click `Yes' in the project recovery window that appears.

**Expected behavior**

Plots should reappear

**Platform/Version (please complete the following information):**
 - OS: All
 - OS Version: N/A
 - Mantid Version 6.12

**Additional context**
Possibly caused by #38486

walshmm, 2025-01-24:
This relates to #38486

walshmm, 2025-01-24:
it is probably that [this line](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/project/plotsloader.py#L111) does not account for the new creation arg

PR #38770 (2025-01-31): Reimplement #38486 to avoid project recovery error
### Description of work

The pr #38486 introduced a problem with project recovery (#38650) by adding a creation arg to all plots. These creating args eventually find themselves passed to an mpl plotting function when being recovered. This causes an error.

The original problem is that when you have a plot that has workspace data (plotted with a workspace name argument) **and** non-workspace data, renaming the plotted workspace threw and exception because the code assumed all sets of plotting arguments would have a workspace name.

I think inst [... cut]
closed 2025-01-31 by github-actions[bot]