mantidproject/mantid #38652: TOF converter should catch division by zero error when two-theta is zero

RichardWaiteSTFC, 2025-01-21:
**Describe the bug**
TOF converter should catch division by zero error when two-theta is zero.
Found during manual testing  #38634

**To Reproduce**
(1) Interfaces > Utility > TOF Converter
(2) Select input unit to be Wavelength, Output to be d-spacing
(3) Set scattering angle to be 0

[image]

**Expected behavior**
Would expect error to be caught and a nicer message like for when input number is 0.

[image]

Note it does actually work for negative angles, just 0 needs to be caught

sf1919, 2025-01-21:
@RichardWaiteSTFC Does this occur in v6.11?

RichardWaiteSTFC, 2025-01-21:
Yes, sorry I mentioned that in the manual testing issue but not here