mantidproject/mantid #38654: mantid launch script opens in the editor on launch

MialLewis, 2025-01-21:
System: OSX
Install: Conda
Version: release-next nightly (manual testing)

When a conda install of mantid is launched using `workbench` or `mantidworkbench` from the terminal, a script of the same name is opened in the editor.

The script contains:
```
#!/bin/bash /Users/wym92549/mambaforge/envs/mantid_612_test/bin/pythonw

# -*- coding: utf-8 -*-
import re
import sys

from workbench.app.mantidworkbench_launch_wrapper import launch

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(launch())

```

**To Reproduce**
1) Install release-next nightly using conda
2) launch workbench using either `workbench` or `mantidworkbench`
3) observe script open in the editor

sf1919, 2025-01-22:
Does this occur in v6.11?

SilkeSchomann, 2025-01-22:
> Does this occur in v6.11?

I will have a look. I'm just installing Mantid 6.11 on MacOS to look for other regressions.

SilkeSchomann, 2025-01-22:
> [@cailafinn](https://github.com/cailafinn) or [@SilkeSchomann](https://github.com/SilkeSchomann), could you see if you can replicate this on OSX?

Yes, I can see the same. It does not happen for Mantid 6.11. It was most likely introduced by this PR: https://github.com/mantidproject/mantid/pull/37750

thomashampson, 2025-01-22:
On Windows the following is printed in the terminal after launching:
`script "C:\Users\dsc49661\AppData\Local\mambaforge\envs\manual_test_6_12\Scripts\workbench" does not exist`

closed 2025-01-23 by github-actions[bot]