mantidproject/mantid #38655: Crash on saving mantid project

GuiMacielPereira, 2025-01-21:
**Describe the bug**
Trying to save mantid project for the first time with `Save altered workspaces only` creates a crash.

**To Reproduce**

- Load `CNCS_7860_event` from training data
- Go to File > Save project
- Tick `Save altered workspaces only`
- Pick a new path for the project (choose a new name)
- Click `OK`

You should be presented with the following error:

```
Traceback (most recent call last):
  File "/home/ljg28444/mambaforge/envs/mantid-6.12-mtest/lib/python3.10/site-packages/mantidqt/widgets/saveprojectdialog/presenter.py", line 66, in save_as
    self.project.save_as(path=self.view.get_location())
  File "/home/ljg28444/mambaforge/envs/mantid-6.12-mtest/lib/python3.10/site-packages/mantidqt/project/project.py", line 109, in save_as
    task.start()
  File "/home/ljg28444/mambaforge/envs/mantid-6.12-mtest/lib/python3.10/site-packages/mantidqt/utils/asynchronous.py", line 148, in start
    raise self.recv.exc_value
  File "/home/ljg28444/mambaforge/envs/mantid-6.12-mtest/lib/python3.10/site-packages/mantidqt/utils/asynchronous.py", line 63, in run
    out = self.target(*self.args, **self.kwargs)
  File "/home/ljg28444/mambaforge/envs/mantid-6.12-mtest/lib/python3.10/site-packages/mantidqt/project/project.py", line 139, in _save
    workspaces_to_save = self._filter_unaltered_workspaces(workspaces_to_save)
  File "/home/ljg28444/mambaforge/envs/mantid-6.12-mtest/lib/python3.10/site-packages/mantidqt/project/project.py", line 194, in _filter_unaltered_workspaces
    if not (history.size() == 1 and history.getAlgorithm(0).name() == "Load"):
RuntimeError: Unknown property search object 'LoadMonitors'
```

This bug was already present in 6.11 and so it is **not a regression**.

**Expected behavior**
The project is saved successfully or at least the error is handled and an error message is displayed.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Ubuntu 20.04
 - OS Version: 20.04
 - Mantid Version 6.12

**Additional context**
@jclarkeSTFC tried it on Windows and managed to also reproduce it.

Despiix, 2025-01-23:
This is also happening in 6.10

jhaigh0, 2025-01-23:
Also in 6.11, you need to tick `LoadMonitors` when loading the data