mantidproject/mantid #38656: Muon analysis UI struggles to find data on archive

RichardWaiteSTFC, 2025-01-21:
**Describe the bug**
Found during manual testing #38633 - but present in v6.11
Seen on IDAaaS and OSX for various instruments

**To Reproduce**

(0) Make sure data archive search enabled in Manage User Directories
(1) Interfaces > Muon > Muon Analysis
(2) Set instrument to ARGUS
(3) Load run `71799` (enter run number and press enter) as in https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_ARGUS.html

[image]

(4) Click `Load Current Run`

[image]

Also see this for other instruments e.g. MUSR on OSX

**Expected behavior**
Would be able to find data on archive - also perhaps this data should live in doc test files?

SilkeSchomann, 2025-01-22:
For MacOS, this happens even more frequently than on other operating systems. This is not a regression from Mantid 6.11, though.

RichardWaiteSTFC, 2025-01-22:
This seems related to more general archive problems (such as time-out) which will be addressed in v6.13 so have changed milestone accordingly

warunawickramasingha, 2026-03-12:
It was seen that after searching in the local paths with possible extensions the [ISISDataArchive.cpp](https://github.com/mantidproject/mantid/blob/b5285c3e41b9768519ca2d2103f33be5c1018d83/Framework/DataHandling/src/ISISDataArchive.cpp#L84) request the location from the `where.py` service. But the expected file does not actually exist in the location returned by that service. Attached following is a log of such scenario from IDAaaS.
[argus_idaas_log.txt](https://github.com/user-attachments/files/25933349/argus_idaas_log.txt)

A similar issue had been reported before as [this issue](https://github.com/mantidproject/mantid/issues/40011). `where.py` seems to return the location of the "current cycle" for unfound files and this was observed by doing some debugging with [Postman](https://www.postman.com/). Therefore it seems the 'fix' for this issue is outside mantid.