mantidproject/mantid #38660: Inconsistent error when closing sliceviewer

MialLewis, 2025-01-22:
**Describe the bug**
During manual testing of 6.12 `release-next` nightly, the following error message was seen when closing sliceviewer:
```
Traceback (most recent call last):  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/matplotlib/cbook.py", line 298, in process    func(*args, **kwargs)  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/widgets/sliceviewer/views/cursor.py", line 57, in on_mouse_leave    self.on_cursor_outside_axes()  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/widgets/sliceviewer/presenters/imageinfowidget.py", line 67, in on_cursor_outside_axes    self._widget.cursorAt(DBLMAX, DBLMAX, DBLMAX, extra_cols)RuntimeError: wrapped C/C++ object of type ImageInfoWidget has been deleted
```

This occurs inconsistently, though was replicated on both IDAaaS and OSX.

s(q,w) portion of https://developer.mantidproject.org/Testing/Inelastic/QENSFittingTests.html

sf1919, 2025-01-22:
Is this a regression?

cailafinn, 2026-01-30:
Adding a comment to potentially un-stale this as we found it again during #40688.

Other (possibly related) error messages:
```py
Traceback (most recent call last):
  File "/home/wj1132075/.conda/envs/manual_testing_6.15/lib/python3.11/site-packages/matplotlib/backend_bases.py", line 2947, in _wait_cursor_for_draw_cm
    self.canvas.set_cursor(tools.Cursors.WAIT)
  File "/home/wj1132075/.conda/envs/manual_testing_6.15/lib/python3.11/site-packages/matplotlib/backends/backend_qt.py", line 267, in set_cursor
    self.setCursor(_api.check_getitem(cursord, cursor=cursor))
RuntimeError: wrapped C/C++ object of type FigureCanvasQTAgg has been deleted

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/wj1132075/.conda/envs/manual_testing_6.15/lib/python3.11/site-packages/matplotlib/cbook.py", line 298, in process
    func(*args, **kwargs)
  File "/home/wj1132075/.conda/envs/manual_testing_6.15/lib/python3.11/site-packages/mantidqt/widgets/sliceviewer/cutviewer/view.py", line 193, in _on_figure_canvas_draw
    self.figure.canvas.draw()
  File "/home/wj1132075/.conda/envs/manual_testing_6.15/lib/python3.11/site-packages/matplotlib/backends/backend_agg.py", line 385, in draw
    with (self.toolbar._wait_cursor_for_draw_cm() if self.toolbar
  File "/home/wj1132075/.conda/envs/manual_testing_6.15/lib/python3.11/contextlib.py", line 137, in __enter__
    return next(self.gen)
           ^^^^^^^^^^^^^^
  File "/home/wj1132075/.conda/envs/manual_testing_6.15/lib/python3.11/site-packages/matplotlib/backend_bases.py", line 2950, in _wait_cursor_for_draw_cm
    self.canvas.set_cursor(self._last_cursor)
  File "/home/wj1132075/.conda/envs/manual_testing_6.15/lib/python3.11/site-packages/matplotlib/backends/backend_qt.py", line 267, in set_cursor
    self.setCursor(_api.check_getitem(cursord, cursor=cursor))
RuntimeError: wrapped C/C++ object of type FigureCanvasQTAgg has been deleted```
```

```py
Exception ignored in: <function WeakMethod.__new__.<locals>._cb at 0x79af8b8e3b00>
Traceback (most recent call last):
  File "/home/sam/software/mantidworkbenchnightly/lib/python3.11/weakref.py", line 60, in _cb
    callback(self)
  File "/home/sam/software/mantidworkbenchnightly/lib/python3.11/site-packages/matplotlib/cbook.py", line 248, in _remove_proxy
    del self.callbacks[signal][cid]
        ~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'changed'
```

jclarkeSTFC, 2026-10-07:
@cailafinn That error I can reproduce by opening SliceViewer, selecting the cutting tool, closing SliceViewer, opening it again, select cutting tool, boom.

closed 2026-10-15 by github-actions[bot]