mantidproject/mantid #38797: Standalone package does not launch on Ubuntu

thomashampson, 2025-02-07:
**Original reporter:** [see here](https://forum.mantidproject.org/t/how-to-install-on-linux/931)

**Describe the bug**
Since v6.10.0, The Linux standalone tarball does not launch on Ubuntu (and probably other Linux flavours). It does launch on IDAaaS. Here is an example of the error that is output:

```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
xkbcommon: ERROR: failed to add default include path /jenkins_workdir/workspace/release-next_nightly_deployment/_bundle_build/mantidworkbench/share/X11/xkb
xkbcommon: ERROR: failed to add default include path /jenkins_workdir/workspace/release-next_nightly_deployment/_bundle_build/mantidworkbench/share/X11/xkb
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
xkbcommon: ERROR: failed to add default include path /jenkins_workdir/workspace/release-next_nightly_deployment/_bundle_build/mantidworkbench/share/X11/xkb
xkbcommon: ERROR: failed to add default include path /jenkins_workdir/workspace/release-next_nightly_deployment/_bundle_build/mantidworkbench/share/X11/xkb
Segmentation fault (core dumped)

```

**To Reproduce**
1. [Download linux standlaone package](https://github.com/mantidproject/mantid/releases/download/v6.11.0/mantidworkbench-6.11.0.tar.xz)
2. Extract package
3. Go into the bin directory
4. attempt to launch with `./mantidworkbench`

**Expected behavior**
Extracting and launching the tarball on Ubuntu should work.

**Platform/Version (please complete the following information):**
 - OS: Ubuntu
 - Mantid Version 6.10.0 onwards.

jclarkeSTFC, 2025-02-11:
Skipping this section from `bin/mantidworkbench` prevents the segmentation fault:
```bash
# Define parameters for jemalloc
LOCAL_PRELOAD=$INSTALLDIR/lib/libjemalloc.so.2
if [ -n "${LD_PRELOAD}" ]; then
    LOCAL_PRELOAD=${LOCAL_PRELOAD}:${LD_PRELOAD}
fi
```
So must be some version incompatibility with newer Ubuntu versions.

thomashampson, 2025-02-11:
It laucnhes fine for some people who have Ubuntu 22.04, but not others.

PR #38819 (2025-02-13): Fix Linux standalone package for Ubuntu
### Description of work
On some Ubuntu systems (does not seem to be version specific, works on some 22.04 but not others), the standalone package fails to launch. This addresses that. I have also renamed the launcher script to `mantidworkbench` for consistency with previous versions.

Fixes #38797 

### To test:
On many Linux flavours (Ubuntu {20, 22, 24}, PopOS, IDAaaS, ...):
- Download `mantidworkbenchunstable-6.11.20250213.0905.tar.xz` from https://builds.mantidproject.org/job/build_packages_from_branch/995/
- Unpack it
- Launch using `./bin/mantidwor [... cut]
closed 2025-02-13 by github-actions[bot]