mantidproject/mantid #38914: Windows Plotting Smoke Tests  :chart_with_upwards_trend: :bar_chart:

MohamedAlmaki, 2025-02-20:
We must test both the .exe installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.12.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
 - [ ] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [ ] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Change colour map 
 - [ ] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [ ] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ74041` from the ISIS sample data
 - [ ] 3D surface
 - [ ] 3D wire frame  
 - [ ] Check Toolbar buttons

 ## Sliceviewer
 - [ ] Overly long instructions (don't spend  long!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer!

RichardWaiteSTFC, 2025-02-24:
Error when changing normalisation with `+` toolbar button enabled

[image]

```
Traceback (most recent call last):
  File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\workbench\plotting\figureinteraction.py", line 499, in <lambda>
    norm_action = norm_menu.addAction("Bin Width", lambda: self._set_normalization_bin_width(ax))
  File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\workbench\plotting\figureinteraction.py", line 804, in _set_normalization_bin_width
    self._toggle_normalization(ax)
  File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\workbench\plotting\figureinteraction.py", line 842, in _toggle_normalization
    self._change_plot_normalization(ax)
  File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\workbench\plotting\figureinteraction.py", line 878, in _change_plot_normalization
    if arg_set["workspaces"] in ax.tracked_workspaces:
KeyError: 'workspaces'
```

RichardWaiteSTFC, 2025-02-24:
Getting a few errors I've seen before - on opening sliceviewer I sometimes get this

```
Traceback (most recent call last):
  File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\matplotlib\cbook.py", line 298, in process
    func(*args, **kwargs)
  File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\mantidqt\widgets\sliceviewer\views\cursor.py", line 57, in on_mouse_leave
    self.on_cursor_outside_axes()
  File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\mantidqt\widgets\sliceviewer\presenters\imageinfowidget.py", line 67, in on_cursor_outside_axes
    self._widget.cursorAt(DBLMAX, DBLMAX, DBLMAX, extra_cols)
RuntimeError: wrapped C/C++ object of type ImageInfoWidget has been deleted
```

On overlaying peaks in 3D MD workspace I get this (but seems to work)
```
Traceback (most recent call last):
 File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\matplotlib\cbook.py", line 298, in process
 func(*args, **kwargs)
 File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\mantidqt\widgets\sliceviewer\views\cursor.py", line 57, in on_mouse_leave
 self.on_cursor_outside_axes()
 File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\mantidqt\widgets\sliceviewer\presenters\imageinfowidget.py", line 66, in on_cursor_outside_axes
 extra_cols = self._presenter.get_extra_image_info_columns(DBLMAX, DBLMAX)
 File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\mantidqt\widgets\sliceviewer\presenters\presenter.py", line 573, in get_extra_image_info_columns
 if len(qdims) != 3 or self.get_frame() != SpecialCoordinateSystem.HKL:
 File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\mantidqt\widgets\sliceviewer\presenters\presenter.py", line 172, in get_frame
 return self.model.get_frame()
 File "C:\Users\xhg73778\AppData\Local\mambaforge\envs\smoketest_mantid_6.12\lib\site-packages\mantidqt\widgets\sliceviewer\models\model.py", line 150, in get_frame
 return self._ws.getSpecialCoordinateSystem()
RuntimeError: Variable invalidated, data has been deleted.
```

closed 2025-02-24 by RichardWaiteSTFC