mantidproject/mantid #38919: MacOS Fitting Smoke Tests  :muscle: :runner:

MohamedAlmaki, 2025-02-20:
We must test both the .dmg installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.12.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

JackEAllen, 2025-02-24:
User Defined Function Drop Down box in Left hand menu for fitting parameters seems to be very small and hard to select the 3 dots to open the larger sub-window for user defined functions. 

[image]

SilkeSchomann, 2025-02-24:
> User Defined Function Drop Down box in Left hand menu for fitting parameters seems to be very small and hard to select the 3 dots to open the larger sub-window for user defined functions.
> 
> [image]

There is an issue for this: https://github.com/mantidproject/mantid/issues/37579

closed 2025-02-24 by JackEAllen