mantidproject/mantid #38925: Linux Clean/Dirty Install Smoke Tests  :soap: :hankey:

MohamedAlmaki, 2025-02-20:
* To install the tar.xz package for Linux, run `(sudo) tar -xJf mantid-VA.B.C.tar.xz` in a terminal and it will unzip the package in your current working directory. 
* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.12.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :soap: :hankey:

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the new release candidate of Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files

**On Windows**:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Remove the mantid  files in %APPDATA%

**On macOS** :
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove (or rename) `~/Library/Preferences/org.python.python.Python.plist`  and `~/Library/Preferences/org.python.python.plist` 

**On Linux** :
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

Re-install the new release candidate of Mantid

- [ ] Check that Mantid boots up correctly

MohamedAlmaki, 2025-02-24:
Tested both clean and dirty installations on my Linux machine using a standalone installer

GuiMacielPereira, 2025-02-24:
Tested both dirty and clean using mamba install

closed 2025-02-24 by GuiMacielPereira