mantidproject/mantid #38964: Update UB matrix if IndexPeaks called with CommonUBForAll=False and table contains only peaks from one run

RichardWaiteSTFC, 2025-02-24:
**Is your feature request related to a problem? Please describe.**
If a peak table contains only peaks from a single run then `IndexPeaks` called with `CommonUBForAll=False` will optimise the UB but this won't be stored on the workspace

https://github.com/mantidproject/mantid/blob/56f61efdf9ca71ef6b64c17c4b9bee8748b462db/Framework/Crystal/src/IndexPeaks.cpp#L503-L506

**Describe the solution you'd like**

Update UB matrix if `IndexPeaks` called with `CommonUBForAll=False` and table contains only peaks from one run - in this way if `RoundHKLs=False`
`peak.getQSampleFrame() = 2*np.pi*peaks.sample().getOrientedLattice().getUB() @ peak.getHKL()`

Why throw away a better UB? It's also better to see which lattice parameters change etc. 

Perhaps for backwards compatibility would require an additonal parameter e.g. `UpdateUB = False` - but it's difficult as this only applie to some epak tables (with peaks from 1 run).

**Describe alternatives you've considered**
Can call `CalculateUMatrix` with indexed peaks and observed lattice parameters, this does seem to produce better results sometimes than e.g. `FindUBUsingLatticeParameters`

**Additional context**
How to treat tables with peaks from multiple runs? Could update the goniometer matrix for peaks from the same run such that
`R_old * UB_new = R_new * UB_old`?
Not sure this is intuitive/desired behaviour though

RichardWaiteSTFC, 2025-02-24:
@zjmorgan what do you think of this? We don't have to do it, but it seems a shame to throw away a better UB!

zjmorgan, 2025-02-24:
Seems like a good idea

PR #39541 (2025-08-07): Add option to update UB matrix when CommonUBForAll=False and table contains only peaks from one run 
### Description of work
This PR adds `UpdateUB` option to `IndexPeaks` which saves the optimized UB matrix to the peaks workspace if `CommonUBForAll=False` and the workspace contains a single run. 

Fixes #38964. 

### To test:
Run the following script:
```
ws = LoadEmptyInstrument(Filename="SXD_Definition.xml", OutputWorkspace="empty_SXD")
axis = ws.getAxis(0)
axis.setUnit("TOF")

bad_ub = np.diag([1/3.8, 0.25, 0.1])  # alatt = [3.8, 4, 10]

peaks = CreatePeaksWorkspace(InstrumentWorkspace=ws, [... cut]
commit 8af461dd (2025-08-08): Add option to update UB matrix when CommonUBForAll=False and table contains only peaks from one run  (#39541)

This PR adds `UpdateUB` option to `IndexPeaks` which saves the optimized
UB matrix to the peaks workspace if `CommonUBForAll=False` and the
workspace contains a single run.

Fixes #38964.

### To test:
Run the following script:
```
ws = LoadEmptyInstrument(Filename="SXD_Definition.xml", OutputWorkspace="empty_SXD")
axis = ws.getAxis(0)
axis.setUnit("TOF")

bad_ub = np.diag([1/3.8, 0.25, 0.1])  # alatt = [3.8, 4, 10]

peaks = CreatePeaksWorkspace(InstrumentWorkspace=ws, NumberOfPeaks=0 [... cut]