mantidproject/mantid #39059: Error opening short MERLIN data file

mducle, 2025-03-14:
**Original reporter:** V. Peçanha-Antonio

**Describe the bug**

Trying to load a short MERLIN run using `Mantid>=6.10` gives the following error:

```
Cannot guess ending value from a TimeSeriesProperty that contains only a single time
```

**To Reproduce**

The problematic datafile is on Babylon: `\\Olympic\Babylon5\Public\duc\MER70731.nxs` (43MB). Load it with Mantid 6.9 works fine. With newer Mantid versions, the above error occurs.

A workaround is to use:

```
LoadEventNexus('MER70731.nxs', LoadLogs=False, OutputWorkspace='MER70731')
LoadNexusMonitors('MER70731.nxs', OutputWorkspace='MER70731_monitors')
CopyLogs('MER70731_monitors', 'MER70731')
```

**Expected behavior**

Able to load the datafile.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Rocky Linux 8 (DAaaS)
 - Mantid Version: 6.12.0

**Additional context**

sf1919, 2025-03-14:
This looks to be similar to an issue we already have open #38598

mducle, 2025-09-05:
@warunawickramasingha Seems someone cleared out Babylon5... I just put the file back.

I also put it in the auxiliary folder on DAaaS at: `/mnt/ceph/auxiliary/excitations/formantid` in case it gets deleted again from Babylon5.

PR #39870 (2025-09-15): Catch runtime errors when adding period logs
### Description of work

Inconsistencies sometimes occur in period-related logs within Event NeXus files, such as duplicated entries with the same timestamps. Therefore, these repeated entries are reduced to a single entry. This can lead to runtime errors when loading the file because it is not possible to determine the ending value from a TimeSeriesProperty with a single time entry. In the past, this has resulted in Mantid crashes. In order to prevent this, the related event workspace is now created without the period logs if issues are detected.

 [... cut]