mantidproject/mantid #39075: Confusing behaviour with Absorption Correction

andy-bridger, 2025-03-17:
**Describe the bug**
Changing sample shape definition from CSG to STL seems to have impact on `AbsorptionCorrection`

**To Reproduce**
Use the following script:
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

gv = " \
<cuboid id='some-gv'> \
<height val='0.004'  /> \
<width val='0.004' />  \
<depth  val='0.004' />  \
<centre x='0.0' y='0.0' z='0.0'  />  \
</cuboid>  \
<algebra val='some-gv' /> \
"

sample = " \
<cuboid id='some-cuboid'> \
<height val='0.1'  /> \
<width val='0.1' />  \
<depth  val='0.1' />  \
<centre x='0.0' y='0.0' z='0.0'  />  \
</cuboid>  \
<algebra val='some-cuboid' /> \
"

ws = CreateSampleWorkspace("Histogram",NumBanks=1,BankPixelWidth=2)
ws = ConvertUnits(ws, Target="Wavelength")

ws2 = CreateSampleWorkspace("Histogram",NumBanks=1,BankPixelWidth=2)
ws2 = ConvertUnits(ws2, Target="Wavelength")

r0 = np.array(([[ 8.66025404e-01, -5.00000000e-01,  5.55111512e-17],
 [-5.55111512e-17, -2.22044605e-16, -1.00000000e+00],
 [ 5.00000000e-01,  8.66025404e-01, -2.22044605e-16]]))
 
r15 = np.array(([[-4.44089210e-16,  2.58819045e-01, -9.65925826e-01],
 [ 1.00000000e+00,  1.11022302e-16, -1.66533454e-16],
 [ 1.66533454e-16, -9.65925826e-01, -2.58819045e-01]]))

R = r0 

# set goniometers
ws.run().getGoniometer().setR(R)
ws2.run().getGoniometer().setR(R)

# set sample either from csg or stl
SetSample(ws, Geometry={'Shape': 'CSG', 'Value': sample})
SetSampleMaterial(ws,ChemicalFormula="Fe")

LoadSampleShape(InputWorkspace=ws2, OutputWorkspace=ws2, Filename= "mesh.stl")
SetSampleMaterial(ws2,ChemicalFormula="Fe")

# def gauge volumes
DefineGaugeVolume(ws, gv)
DefineGaugeVolume(ws2, gv)

# run corrections
abscor = AbsorptionCorrection(ws, ElementSize = 1.0)
corr = ws / abscor
abscor2 = AbsorptionCorrection(ws2, ElementSize = 1.0)
corr2 = ws2 / abscor2
```
`mesh.stl` is a cuboid equivalent to the CSG cuboid (looking at `Show Sample Shape` for `ws` and `ws2` should show they are the same). r0 and r15 are rotation matrices. Try swapping around r0, r15 and looking at `ws`, `ws2`, `abscor`, `abscor2`, `corr` and `corr2` which should be equivalent.

**Expected behavior**
I would expect the corrections in the `r0`, `corr2` case (below) to look like the others (further below)

[image]

Very different results for all the other cases:

`r0`, `corr`:

[image]

`r15`:

[image]

At the least behaviour should have `corr == corr2` and `r0` data and `r15` data should probably be of the same order of magnitude?

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

RichardWaiteSTFC, 2025-03-17:
Do you still see this discrepancy without the gauge volume?
How does this compare to the results of `MonteCarloAbsorption`?

RichardWaiteSTFC, 2025-03-18:
Also you could try comparing to the Dwiggins tables for a sphere - as I do in this PR for `MonteCarloAbsoprtion` #32922 - cheers!

andy-bridger, 2025-03-18:
With no gauge volume, it appears more sensible...
The r0 rotations now match, but the r15 have intensity discrepancies

[image]

andy-bridger, 2025-03-18:
Both are of similar magnitudes to the monte carlo approach now, so I think it is probably a sample shape issue?

[image]

andy-bridger, 2025-03-18:
> Also you could try comparing to the Dwiggins tables for a sphere - as I do in this PR for `MonteCarloAbsoprtion` [#32922](https://github.com/mantidproject/mantid/pull/32922) - cheers!

With the CSG, and an ElementSize of 1mm for the 1cm sphere, gives a value within 0.05% of the Dwiggins value.

andy-bridger, 2025-03-18:
Definitely a problem using the `stl`:

[image]

RichardWaiteSTFC, 2025-03-18:
Thanks for checking all this. Does the result depend on the mesh density in the .stl? Is the .stl mesh filling the sample volume?

andy-bridger, 2025-03-18:
I'm just using a generic mesh library, so I am not certain, but it is probably just a surface mesh so that could well be the problem. I'll have a look at converting them to volumetric meshes. 

If that is the problem, the behaviour being rotation dependent is still somewhat interesting, I would expect the behaviour to be: when the mesh is rasterised into voxels, if it is a surface mesh, most of the voxels are thrown away as 'not within the sample' unless they directly intercept a surface face. 

But I wouldn't expect any of the voxels within the central 4mm^3 gauge volume of a 1cm^3 cube to intercept a face, regardless of the orientation?

RichardWaiteSTFC, 2025-03-18:
I suppose if the mesh is on the surface only and the cuboid mesh in `AbsorbtionCorrection` has axes aligned to the lab frame, when you rotate, the effective thickness of the shell is changing? I think the first step is to compare a volume filled mesh and CSG shape that are equivalent. Perhaps easier said than done though!

andy-bridger, 2025-03-18:
From what I gather, `stl` files only ever contain the surface geometry (https://en.wikipedia.org/wiki/STL_(file_format)) so if the algorithm does expect volumetric meshes that is a problem. I don't think it does, it seems to ray trace intersections and determine entries and exits based on the ordering of points being anticlockwise if viewed from outside the shape. I think this is probably the assumption which is failing, but I'll check my file. I suspect that certain faces are defined ok, but others aren't and that's where the rotational dependence is coming from.

RichardWaiteSTFC, 2025-03-18:
OOK sounds good - thanks again for looking into it! Maybe we need an additional method/function to sort the faces/points when the sample shape is set?

andy-bridger, 2025-03-18:
I'm not sure it can be done automatically and it does say on the `LoadSampleShape` docs, so this might just need to be chalked up to user error.

andy-bridger, 2025-03-18:
Remembered that there is still discrepancy with the gauge volume to work out...

```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

def cuboid_shape(l, id):
   return f" \
<cuboid id='{id}'> \
<height val='{l}'  /> \
<width val='{l}' />  \
<depth  val='{l}' />  \
<centre x='0.0' y='0.0' z='0.0'  />  \
</cuboid>  \
<algebra val='{id}' /> \
"

sample = cuboid_shape(0.1, "sample")
gv_equal = cuboid_shape(0.1, "eq_gv")
gv_bigger = cuboid_shape(0.18, "big_gv")
gv_small = cuboid_shape(0.005, "small_gv")

ws = CreateSampleWorkspace("Histogram",NumBanks=1,BankPixelWidth=2)
ws = ConvertUnits(ws, Target="Wavelength")
SetSample(ws, Geometry={'Shape': 'CSG', 'Value': sample})
SetSampleMaterial(ws,ChemicalFormula="Fe")

ws2 = CloneWorkspace(ws)
ws3 = CloneWorkspace(ws)
ws4 = CloneWorkspace(ws)

# def gauge volumes
DefineGaugeVolume(ws2, gv_equal)
DefineGaugeVolume(ws3, gv_bigger)
DefineGaugeVolume(ws4, gv_small)

# run corrections
abs_corr_no_gv = AbsorptionCorrection(ws, ElementSize = 1.0)
abs_corr_eq_gv = AbsorptionCorrection(ws2, ElementSize = 1.0)
abs_corr_big_gv = AbsorptionCorrection(ws3, ElementSize = 1.0)
abs_corr_small_gv = AbsorptionCorrection(ws4, ElementSize = 1.0)

abs_mc = MonteCarloAbsorption(ws2)

```

[image]

andy-bridger, 2025-03-19:
there might be a problem with `Rasterize`

```
  // go through the bounding box generating cubes and seeing if they are
  // inside the shape
  for (size_t i = 0; i < numZSlices; ++i) {
    const double z = (static_cast<double>(i) + 0.5) * ZSliceThickness + bbox.xMin();

    for (size_t j = 0; j < numYSlices; ++j) {
      const double y = (static_cast<double>(j) + 0.5) * YSliceThickness + bbox.yMin();

      for (size_t k = 0; k < numXSlices; ++k) {
        const double x = (static_cast<double>(k) + 0.5) * XSliceThickness + bbox.zMin();
        // Set the current position in the sample in Cartesian coordinates.
        const Kernel::V3D currentPosition = V3D(x, y, z);
        // Check if the current point is within the object. If not, skip.
        if (shape.isValid(currentPosition)) {
          // Create track for distance in sample before scattering point
          Track incoming(currentPosition, -beamDirection);
          // We have an issue where occasionally, even though a point is
          // within the object a track segment to the surface isn't correctly
          // created. In the context of this algorithm I think it's safe to
          // just chuck away the element in this case. This will also throw
          // away points that are inside a gauge volume but outside the sample
          if (shape.interceptSurface(incoming) > 0) {
            result.l1.emplace_back(incoming.totalDistInsideObject());
            result.position.emplace_back(currentPosition);
            result.volume.emplace_back(elementVolume);
          }
        }
      }
    }
  }

  // Record the number of elements we ended up with
  result.totalvolume = static_cast<double>(result.l1.size()) * elementVolume;
``` 

For the gauge volume greater than the sample size, this `result.totalvolume` is the entire gauge volume, suggesting that "This will also throw away points that are inside a gauge volume but outside the sample" is not what is happening?

The small gauge vol seems to be working correctly, at least at this rasterisation stage

andy-bridger, 2025-03-19:
I think the problem is that `auto raster = Geometry::Rasterize::calculate(m_beamDirection, *integrationVolume, m_cubeSide);` only uses the integration volume for the calculations and so basically ends up assuming the full gauge volume is valid as the shape each point is checked to be within is the gauge volume shape rather than the sample shape?

closed 2025-03-19 by andy-bridger