mantidproject/mantid #39088: LoadCIF parsing short form spacegroup symbol

mducle, 2025-03-21:
**Is your feature request related to a problem? Please describe.**

The `LoadCIF` algorithm in Mantid depends on an internal parser which uses C++ classes in the `Geometry` library. In particular, the `Geometry::SpaceGroupFactory` [class](https://github.com/mantidproject/mantid/blob/77b9bd77d265fe17b6c21ea6a3250abc31e686bd/Framework/Geometry/src/Crystal/SpaceGroupFactory.cpp#L454) only accepts **full** Hermann-Mauguin (H-M, or "international") symbols. However, there are also *short* H-M symbols as [described here](https://dictionary.iucr.org/Hermann-Mauguin_symbols):

> Short and full Hermann-Mauguin symbols coincide for most space groups with the exception of the monoclinic space groups (the full symbol explicitly states the symmetry direction: for example, Pc short symbol vs P1c1 full symbol)

Unfortunately, the `cif` produced by some popular programs such as Vesta use short symbols and this is not recognised by Mantid.

**Describe the solution you'd like**

Add aliases to short H-M symbols in [Geometry/Crystal/SpaceGroupFactory.cpp](https://github.com/mantidproject/mantid/blob/77b9bd77d265fe17b6c21ea6a3250abc31e686bd/Framework/Geometry/src/Crystal/SpaceGroupFactory.cpp#L454) so that short symbols can be understood by `LoadCIF`.

**Describe alternatives you've considered**

A work-around is to manually edit the `cif` to replace the short with the full symbols (e.g. add `1` for the axes being omitted in the short symbol). 

To facilitate this it might be helpful to amend the [error message](https://github.com/mantidproject/mantid/blob/77b9bd77d265fe17b6c21ea6a3250abc31e686bd/Framework/PythonInterface/plugins/algorithms/LoadCIF.py#L64-L68) to note that Mantid only understands *full* H-M symbols.

**Additional context**

For monoclinic spacegroups there is only one unique axis which has a symmetry element which is not the identity (denoted by `1`). In the full H-M, the non-unique axes are denoted by `1` but these are omitted by the short symbols. E.g. `C 1 2/c 1` is the full symbol but `C 2/c` is the short symbol, indicating the unique axis is the `b` axis (middle element).

mducle, 2025-03-24:
An example file with short H-M symbol:

[TbBO3V.cif.txt](https://github.com/user-attachments/files/19427738/TbBO3V.cif.txt)

If you replace the `C 2/c`  in line 15 with `C 1 2/c 1` then LoadCIF will accept it.

SilkeSchomann, 2025-03-27:
@mducle I have seen two options for short H-M symbols. One has a space between the element and the unique axis, the other one doesn't. For instance, `C 2/c` and `C2/c`. Is it safer to consider both?

mducle, 2025-03-27:
@SilkeSchomann I think that strictly the version without space is incorrect - the H-M symbols are meant to be made up of 4 components: The centering (`P`rimitive, `A`-, `B`, `C`-face-centered, `F`ace-centered, `I` body-centered) and then three symbols indicating the highest rotation/mirror/glide symmetry in each of the three directions in the crystal. In principle there should be a space between each of the four component. However, often people omit the spaces (especially in papers!) as it is relatively easy for a person with some knowledge of crystallography to parse the symbol without the spaces.

I think the best option would be to use this table from spglib:

https://github.com/spglib/spglib/blob/0218486cbf747cf1984ae8ccc2edb1cc6c1ff9a1/database/spg.csv#L57

where column 8 (with `<symbol> = <symbol>`) has both the short and long-form symbols. (Column 5 is the index of the space group in the International Tables for Crystallography).

If you want you can just store the symbols without spaces and then have a code remove spaces from the inputs since we don't actually do any further processing with the different symmetry elements (I think!).

closed 2025-04-03 by RichardWaiteSTFC

SilkeSchomann, 2025-05-29:
No, I just noticed that it was done for 6.13, but I had accidentally changed the milestone for 6.14. Same story for https://github.com/mantidproject/mantid/issues/38671

PR #39116 (2025-04-03): Short forms of H-M symbols can now be used in CIF files
### Description of work

#### Summary of work
LoadCIF now accepts the short forms of the Hermann-Mauguin symbols. These short forms may include a single space or none at all.

Fixes #39088.

**Report to:** Duc Le

#### Further detail of work
When parsing a CIF file, short forms of the Hermann-Mauguin symbols are converted into the corresponding long forms.

### To test:

Use the following script for testing:

`from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

sample_ws = CreateSampleWorkspace()
LoadCIF(sam [... cut]