mantidproject/mantid #39110: Mantid crashed with the instrument view and rubber tool

warunawickramasingha, 2025-03-26:
**Original reporter:** [[pascal.manuel@stfc.ac.uk](mailto:pascal.manuel@stfc.ac.uk)]

**Describe the bug**
Mantid crashes when the rubber tool is applied in the instrument view on a WISH runs overlayed with peak workspaces.

**To Reproduce**
(1)	Run the below script
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

from mantid.simpleapi import *

Load(Filename='/archive/NDXWISH/Instrument/data/cycle_18_3/WISH00042732.raw', OutputWorkspace='WISH00042732')
CropWorkspace(InputWorkspace='WISH00042732', OutputWorkspace='WISH00042732', XMin=6000, XMax=99000)
ConvertUnits(InputWorkspace='WISH00042732', OutputWorkspace='WISH00042732', Target='dSpacing')
FindSXPeaks(InputWorkspace='WISH00042732', PeakFindingStrategy='AllPeaks', AbsoluteBackground=10, ResolutionStrategy='AbsoluteResolution', XResolution=0.10000000000000001, PhiResolution=2, TwoThetaResolution=2, OutputWorkspace='peaks')
FindSXPeaks(InputWorkspace='WISH00042732', PeakFindingStrategy='AllPeaks', AbsoluteBackground=8, ResolutionStrategy='AbsoluteResolution', XResolution=0.10000000000000001, PhiResolution=2, TwoThetaResolution=2, OutputWorkspace='peaks2')
```

(2)	Open WISH workspace in instrument view
(3)	Overlay both `peaks` and `peaks2` workspaces – this takes a while…
(4)	Use the rubber tool to start removing peaks
(5)	Mantid hangs then crashes

**Expected behavior**
Mantid should not crash and subjected peaks should be removed successfuly

**Platform/Version (please complete the following information):**
 - OS: [Windows 10, Rocky Linux 8.10]
 - Mantid Version [6.12.0]

sf1919, 2025-04-01:
Did this happen on v6.12 or v6.11? The original report this links to suggests v6.11.

warunawickramasingha, 2025-04-01:
> Did this happen on v6.12 or v6.11? The original report this links to suggests v6.11.

The original issue was reported in v6.11, but this can be created in v6.12 as well.