mantidproject/mantid #39148: Investigate error calling LoadCIF for CEO2.cif

RichardWaiteSTFC, 2025-04-03:
**Describe the bug**
When loading a CIF file (actually using GSAS-II tab) got this error.

```
LoadCIF started
Incorrect number of loop values for loop containing ['_atom_site_type_symbol', '_atom_site_label', '_atom_site_fract_x', '_atom_site_fract_y', '_atom_site_fract_z', '_atom_site_occupancy', '_atom_site_thermal_displace_type', '_atom_site_U_iso_or_equiv', '_atom_site_symmetry_multiplicity']
Unexpected error:
Traceback (most recent call last):
  File "/home/rw1061637/.conda/envs/mantid-test/lib/python3.10/site-packages/CifFile/YappsStarParser_1_1.py", line 80, in makeloop
    target_block.CreateLoop(loop_seq)  #will raise ValueError on problem
  File "/home/rw1061637/.conda/envs/mantid-test/lib/python3.10/site-packages/CifFile/StarFile.py", line 771, in CreateLoop
    raise ValueError('Request to loop datanames %s with different lengths: %s' % (repr( datanames ),repr( len_set )))
ValueError: Request to loop datanames ['_atom_site_type_symbol', '_atom_site_label', '_atom_site_fract_x', '_atom_site_fract_y', '_atom_site_fract_z', '_atom_site_occupancy', '_atom_site_thermal_displace_type', '_atom_site_U_iso_or_equiv', '_atom_site_symmetry_multiplicity'] with different lengths: {1, 2}
```

It could be an issue with the cif file (the error is coming from a third-party library, but it could also be a bug or perhaps we could do some pre-processing to handle this.

**To Reproduce**
```
ws=CreateSampleWorkspace()
LoadCIF(ws, "C:\mantid-conda\mantid\scripts\Engineering\ENGINX\phase_info\CEO2.cif")
```

**Expected behavior**
No crash

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

RichardWaiteSTFC, 2025-04-03:
If see a problem with .cif could you update file (or a new one) and update the manual testing to use this in GSAS tests - check data iused is ceria...thanks!

JackEAllen, 2025-04-30:
After some further exploration, the issue here is possibly caused by how `CEO2.cif` was created. A loop error occurs where the number of loop elements is not a multiple of packetsize at line 449 due to missing/incorrectly formatted keys.

PR #39265 (2025-05-14): 39148 investigate load cif unexpected error
### Description of work

#### Summary of work

Correct possible formatting issue with `CEO2.cif` causing a loop error

The possible formatting error prevented `LoadCIF` from working as intended, outputting the error described within: #39148 

Fixes #39148  

#### Further detail of work

### To test:

1. Try calling loading `CEO2.cif` which can be found at `scripts/Engineering/ENGINX/phase_info/CEO2.cif` within the Mantid Workbench script editor with `LoadCIF`
```Python
ws=CreateSampleWo
commit e05225e2 (2025-05-14): 39148 investigate load cif unexpected error (#39265)

### Description of work

#### Summary of work

Correct possible formatting issue with `CEO2.cif` causing a loop error

The possible formatting error prevented `LoadCIF` from working as
intended, outputting the error described within: #39148

Fixes #39148 

#### Further detail of work

### To test:

1. Try calling loading `CEO2.cif` which can be found at
`scripts/Engineering/ENGINX/phase_info/CEO2.cif` within the Mantid
Workbench script editor with `LoadCIF`
```Python
ws=CreateSampleWorkspace()
LoadCIF(ws, "absolute
closed 2025-05-14 by RichardWaiteSTFC