mantidproject/mantid #39237: Make mslice an optional dependency of mantidworkbench

thomashampson, 2025-04-24:
Turn mslice into an optional dependency of mantidworkbench.

Still ensure that it is always installed:
- in the mantid-developer metapackage
- on IDAaaS
- inside the standalone packages

MialLewis, 2025-04-24:
Do you think it's a good idea to have a metapackage that installs mantid and all of its optional subpackages, replicating legacy behaviour?

thomashampson, 2025-05-09:
> Do you think it's a good idea to have a metapackage that installs mantid and all of its optional subpackages, replicating legacy behaviour?

I think the user would have to specify all the channels when installing this metapackage. For mslice this is not really an issue since it's on the same channel, but there are some optional dependencies that are not on the mantid or conda-forge channels.

closed 2025-07-23 by SilkeSchomann