mantidproject/mantid #39327: Copy to clipboard exception on IDAaaS

jclarkeSTFC, 2025-05-15:
**Describe the bug**

Launch Mantid standalone on IDAaaS, go back to the terminal, highlight some text, press Ctrl+C.
This error in Mantid:
```
Traceback (most recent call last):
  File "/mnt/ceph/home/c1128873/Downloads/mantidworkbenchunstable/lib/python3.11/site-packages/mantidqt/widgets/memorywidget/memoryview.py", line 98, in _set_value
    @Slot(int, float, float)
```
This error in the terminal:
```
^CTraceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/mnt/ceph/home/c1128873/Downloads/mantidworkbenchunstable/lib/python3.11/site-packages/workbench/app/main.py", line 67, in main
    start(options)
  File "/mnt/ceph/home/c1128873/Downloads/mantidworkbenchunstable/lib/python3.11/site-packages/workbench/app/main.py", line 95, in start
    start(options)
  File "/mnt/ceph/home/c1128873/Downloads/mantidworkbenchunstable/lib/python3.11/site-packages/workbench/app/start.py", line 84, in start
    workbench_process.wait()
  File "/mnt/ceph/home/c1128873/Downloads/mantidworkbenchunstable/lib/python3.11/subprocess.py", line 1264, in wait
    return self._wait(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/ceph/home/c1128873/Downloads/mantidworkbenchunstable/lib/python3.11/subprocess.py", line 2053, in _wait
    (pid, sts) = self._try_wait(0)
                 ^^^^^^^^^^^^^^^^^
  File "/mnt/ceph/home/c1128873/Downloads/mantidworkbenchunstable/lib/python3.11/subprocess.py", line 2011, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

```

**Platform/Version (please complete the following information):**
 - OS: Rocky
 - OS Version: IDAaaS
 - Mantid Version [e.g. 6.0.0]

jhaigh0, 2025-05-19:
Is this not expected for a bash shell? I thought `ctrl+c` kills all the shell processes (https://medium.com/@aantipov/what-happens-when-you-ctrl-c-in-the-terminal-36b093443e06). I think I'd either right-click copy or try `ctrl+ins`

jclarkeSTFC, 2025-05-19:
Haha yes good point

closed 2025-05-19 by jclarkeSTFC