mantidproject/mantid #39369: Manual Testing Sliceviewer

MialLewis, 2025-05-23:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html

jhaigh0, 2025-06-03:
`Change the integration-width/slice-thickness (spinbox to the left of the word thick) along the non-viewed axes.`

If I increase it with the arrow, the colour bar number labels don't update, and you end up pushing them off the screen.

GuiMacielPereira, 2025-06-03:
> Check the slicepoint is updated in the cursor info table (note if Track Cursor is checked you will need to click on the colorfill plot again).

I can't see this slicepoint in the cursor info table?

GuiMacielPereira, 2025-06-03:
Took ~1h 20mins

GuiMacielPereira, 2025-06-04:
Found https://github.com/mantidproject/mantid/issues/39447 when testing the slider for overlay peaks