mantidproject/mantid #39370: Manual Testing Engineering Diffraction

MialLewis, 2025-05-23:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html

warunawickramasingha, 2025-06-03:
There was a crash from the Engineering diffraction GUI.

**Precondition**: There were no data in the ADS before following Engineering diffraction test 9
followed the steps and crashed when clicked on Sequantial fit at step 5.

SilkeSchomann, 2025-06-03:
Please add a note to the manual testing instructions that it makes sense to download the test data separately first. It does save a lot of time during testing!

warunawickramasingha, 2025-06-03:
Test time=1h given the required ENGINX files were added to User directories.

warunawickramasingha, 2025-06-06:
### Not a regression
[Another crash](https://github.com/mantidproject/mantid/issues/39465) was identified while testing the GSASII interface on IDAaaS

warunawickramasingha, 2025-06-09:
Tested on Windows with a newer version(5782) of gsas2.
Closing this since all the issues have been raised.