mantidproject/mantid #39371: Manual Testing Muon

MialLewis, 2025-05-23:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

See instructions at: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html

jclarkeSTFC, 2025-06-03:
macOS, ALC testing, set instrument to HIFI, runs to 134028-39, set `1` and `64!` for the forward and backward boxes, click `Load`, Mantid disappears with an unhandled `std::invalid_argument` exception, it can't handle the ! in the backward box.

RichardWaiteSTFC, 2025-06-04:
ALC peak fitting - it's not in the instructions but importing results doesn't seem to work. It populates the data but not the baseline or fitted peak (but maybe that's expected?)

MohamedAlmaki, 2025-06-04:
How did you load the data? I couldn't load any ARGUS files

RichardWaiteSTFC, 2025-06-04:
Finished ~ 2 hours. Don't think there's any issues that need raising

MohamedAlmaki, 2025-06-04:
> but cant reproduce again - did anyone else see this?
I get an error that no function or data is selected when trying to do the sequential fit