mantidproject/mantid #39373: Manual Testing ISIS Reflectometry

MialLewis, 2025-05-23:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

warunawickramasingha, 2025-06-04:
### Not a regression 

While following the [Polarisation Corrections](https://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html#id10) test, at step 7 when `Polarization Efficiencies` line edit is entered with invalid values like `//sdsdsd//` or `\\dsdssds\\` several times, the UI hangs and ultimately **crashes**. This was observed in Windows. 

However, this is **not a regression** as the same happens in Mantid v6.12 as well.

warunawickramasingha, 2025-06-04:
> ### Not a regression
> While following the [Polarisation Corrections](https://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html#id10) test, at step 7 when `Polarization [...]

#39452 was raised for this issue.

closed 2025-06-04 by warunawickramasingha