mantidproject/mantid #39374: Manual Testing Inelastic interfaces

MialLewis, 2025-05-23:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Follow the directions at:

* https://developer.mantidproject.org/Testing/Inelastic/BayesFittingTests.html
* https://developer.mantidproject.org/Testing/Inelastic/CorrectionsTests.html
* https://developer.mantidproject.org/Testing/Inelastic/QENSFittingTests.html 
* https://developer.mantidproject.org/Testing/Inelastic/DataProcessorTests.html

thomashampson, 2025-06-03:
The first time you move the slider onthe preview plot you get this warning message:
`In future, it will be an error for 'np.bool' scalars to be interpreted as an index`
This is not rpesent in 6.12.0 and is therefore a **regression**, most likely due to the Python 3.10 to 3.11 update.

thomashampson, 2025-06-03:
In the [Moments tab](https://developer.mantidproject.org/Testing/Inelastic/DataProcessorTests.html#indirect-data) of the Data Processor interface, it states in step 5:
`First spectrum should be rendered on the plot widget, with two vertical sliders placed on the integration limits, the limits on the sliders and on the property browser should match.`
However, the values do not match until you touch the sliders on the plot:

[image]

### This is not a regression.

thomashampson, 2025-06-04:
All issues written up.

closed 2025-06-04 by thomashampson