mantidproject/mantid #39375: Manual Testing ISIS SANS

MialLewis, 2025-05-23:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html

thomashampson, 2025-06-03:
It is possible to save a csv that will not be loadable. I think it should prevent you from saving a csv that cannot be loaded back in.

jhaigh0, 2025-06-03:
First processing test
`In the workspaces list, there should be a series of new workspaces; four group workspaces and four 1D workspaces.`
I get 5 group workspaces, 4 1D

jhaigh0, 2025-06-03:
Took ~1h 10mins

GuiMacielPereira, 2025-06-03:
Took ~1h 5mins

thomashampson, 2025-06-03:
On Windows, when following the _Scaled Background Subtracted Reduction_ instructions, I get the following error:
```
The two workspaces are not compatible because they have different units for the data (Y).
The two workspaces are not compatible for algorithm Minus
Error in execution of algorithm Minus:
The two workspaces are not compatible for algorithm Minus
```

thomashampson, 2025-06-04:
> I observed no difference when setting Phi Limit to 0 to 45 in the Display Mask test.

Did look like a bug that was present only on Linux?

jhaigh0, 2025-06-05:
Several others have tested the 'bug' I found with setting the phi limits and couldn't replicate