mantidproject/mantid #39592: Plot crosshair interferes with normalisation

jhaigh0, 2025-06-27:
**Describe the bug**

- load some data and plot a spectrum
- right-click to change the normalisation (should work fine)
- turn on the crosshair,
  - right-click and change normalisation, get an error from `figureinteraction.py`
- turn off the crosshair and the error persists. (happens even if you just turn it on and off before changing the normalisation).

Found during smoke testing https://github.com/mantidproject/mantid/issues/39545

**Expected behavior**
The crosshair toggle should have no effect on any other part of the plotting

**Platform/Version (please complete the following information):**
 - OS: Windows
 - OS Version: 11
 - Mantid Version 6.13.0rc1

cailafinn, 2025-06-27:
Error also occurs when hiding errors on a plot. Different stack trace, but the same cause of a missing key for `workspaces`:
```python
  File "/Users/caila.finn/miniforge3/envs/mantid_6.13_smoke/lib/python3.11/site-packages/workbench/plotting/figureerrorsmanager.py", line 90, in update_plot_after
    func(*args, **kwargs)
  File "/Users/caila.finn/miniforge3/envs/mantid_6.13_smoke/lib/python3.11/site-packages/workbench/plotting/figureerrorsmanager.py", line 32, in toggle_all_errors
    cls.toggle_error_bars_for(ax, line, make_visible)
  File "/Users/caila.finn/miniforge3/envs/mantid_6.13_smoke/lib/python3.11/site-packages/workbench/plotting/figureerrorsmanager.py", line 51, in toggle_error_bars_for
    new_curve = cls.replot_curve(ax, curve, plot_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/caila.finn/miniforge3/envs/mantid_6.13_smoke/lib/python3.11/site-packages/workbench/plotting/figureerrorsmanager.py", line 126, in replot_curve
    new_curve = ax.replot_artist(curve, errorbars=curve_has_errors(curve), **plot_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/caila.finn/miniforge3/envs/mantid_6.13_smoke/lib/python3.11/site-packages/mantid/plots/mantidaxes.py", line 535, in replot_artist
    self._remove_matching_curve_from_creation_args(workspace.name(), workspace_index, spec_num)
  File "/Users/caila.finn/miniforge3/envs/mantid_6.13_smoke/lib/python3.11/site-packages/mantid/plots/mantidaxes.py", line 1328, in _remove_matching_curve_from_creation_args
    if workspace_name == creation_arg["workspaces"]:
                         ~~~~~~~~~~~~^^^^^^^^^^^^^^
KeyError: 'workspaces'
```

cailafinn, 2025-06-27:
At a quick look, this is down to the crosshairs being plotted as curves on the MantidAxes object, but without being connected to a workspace, like other curves are expected to be. 

A fix may be plotting the crosshair in the same way we plot the markers?

cailafinn, 2025-06-27:
> Probably worth sticking this here: Could we change the icon on the plotting GUI from a + to something that is more clearly a crosshair (+ in a circle)

https://pictogrammers.com/library/mdi/icon/crosshairs/ May be an option

MialLewis, 2025-06-27:
There is a discussion about this here: https://matplotlib.org/stable/gallery/event_handling/cursor_demo.html

Obviously out of scope of this release!

PR #39598 (2025-06-27): Crosshair bugfix release next
### Description of work
This PR handles errors that arises from crosshair functions interacting with mantid. PR [39491](https://github.com/mantidproject/mantid/pull/39491)

The issue is when changing normalisation and error bars, the figure object is treated as mantid object such as mantidaxes with key word arg such as "workspaces". The crosshair, at the moment, is implemented purely as matplotlib object and would cause error and crash the plot. A quick fix is to catch these errors.

In the long term,  we will integrate the crosshair as a mantid specific object fo [... cut]
commit 6a5ee26a (2025-06-27): Crosshair bugfix release next (#39598)

### Description of work
This PR handles errors that arises from crosshair functions interacting
with mantid. PR
[39491](https://github.com/mantidproject/mantid/pull/39491)

The issue is when changing normalisation and error bars, the figure
object is treated as mantid object such as mantidaxes with key word arg
such as "workspaces". The crosshair, at the moment, is implemented
purely as matplotlib object and would cause error and crash the plot. A
quick fix is to catch these errors.

In the long term, we will integrate the crosshair as a mantid specific
 [... cut]
closed 2025-06-27 by mantid-builder