mantidproject/mantid #39954: Performance regression for LoadMD/SaveMD

AndreiSavici, 2025-09-18:
**Describe the bug**
LoadMD/SaveMD slowed down by a factor of 30 between 6.12 and 6.13.1.2

**To Reproduce**
Run the following script in 6.12 and 6.13.1.2

```
from mantid.simpleapi import *
import mantid
import time
import os

output_filename = f'/SNS/users/3y9/Desktop/dm1_sliced_{mantid.__version__}.nxs'
LoadMD(Filename='/SNS/users/3y9/Desktop/dm1.nxs', OutputWorkspace='dm1')
SliceMD(InputWorkspace='dm1', 
        AlignedDim0='Q_sample_x, -13,13,5', 
        AlignedDim1='Q_sample_y, -13,13,5', 
        AlignedDim2='Q_sample_z, -13,13,5', 
        AlignedDim3='DeltaE, -57,57.1,5', 
        OutputWorkspace='w_new_slice', 
        TakeMaxRecursionDepthFromInput=False, 
        MaxRecursionDepth=1)
start_time = time.time()
SaveMD(InputWorkspace='w_new_slice', Filename=output_filename)
save_time = time.time()
elapsed_time1 = save_time - start_time
print(f"Elapsed time saving: {elapsed_time1:.6f} seconds")
new_ws = LoadMD(Filename=output_filename)
load_time = time.time()
elapsed_time2 = load_time - save_time
print(f"Elapsed time loading: {elapsed_time2:.6f} seconds")
size_in_MB = os.path.getsize(output_filename)/(1024*1024)
print(f"Saved file size: {size_in_MB:.1f} MB")
```

I have the following output:
```
6.12.0

SliceMD started
164527436 MDEvents added to the output workspace.
SliceMD successful, Duration 15.50 seconds
SaveMD started
SaveMD successful, Duration 41.53 seconds
Elapsed time saving: 41.535187 seconds
LoadMD started
LoadMD successful, Duration 35.76 seconds
Elapsed time loading: 35.761759 seconds
Saved file size: 6674.3 MB

6.13.1.2 (memory usage >400 GB then goes down to 11 GB)

SliceMD successful, Duration 17.32 seconds
SaveMD started
SaveMD successful, Duration 21 minutes 21.17 seconds
Elapsed time saving: 1281.178254 seconds
LoadMD started
LoadMD successful, Duration 21 minutes 47.60 seconds
Elapsed time loading: 1307.619143 seconds
Saved file size: 14002.3 MB
```

Loading the 6.12 file in 6.13.1.2 is fast

**Expected behavior**
Comparable loading/saving times

**Platform/Version (please complete the following information):**
 - OS: RHEL9
 - OS Version:
 - Mantid Version 6.13.1.2

**Additional context**

rboston628, 2025-09-29:
I just tried on 6.13.1.  Ran the following script to test time scaling:
``` python
from mantid.simpleapi import *
import mantid
import time
import os
import matplotlib.pyplot as plt

sizes = ['4.64527e+05', '1.64527e+06', '4.64527e+06', '1.64527e+07', '4.64527e+07', '1.64527e+08']
save_times = []
load_times = []
workspace_sizes = []
memories = []

kbToGB = 1024 * 1024

for size in sizes:
    CreateMDWorkspace(
        Dimensions=4,
        Extents='-10,10,-10,10,-10,10,-10,10',
        Names='a,b,c,d', Units='rlu,rlu,rlu,meV',
        MaxRecursionDepth=1,
        OutputWorkspace='w_new_slice')
    FakeMDEventData(InputWorkspace='w_new_slice', UniformParams=size)

    output_filename = f'~/Documents/dm1_sliced_{mantid.__version__}.nxs'
    start_time = time.time()
    SaveMD(InputWorkspace='w_new_slice', Filename=output_filename)
    save_times.append(time.time() - start_time)
    workspace_sizes.append(mtd["w_new_slice"].getMemorySize()/kbToGB)
    start_time = time.time()
    LoadMD(Filename=output_filename, OutputWorkspace = "reloaded_ws")
    load_times.append(time.time() - start_time)
    memories.append(os.path.getsize(os.path.expanduser(output_filename))/kbToGB)
```

Plotting the times, I got the following:
[image]

For the ~4GB file, saing and loading was ~1s.  I think this shows the error was introduced *after* 6.13.1.

rboston628, 2025-09-29:
Since compression is a suspect, I looked at `SaveNexusProcessed`, which also used compression.  Ran the following:
``` python
from mantid.simpleapi import *
import mantid
import time
import os

CreateSampleWorkspace(
    OutputWorkspace="new_ws",
    WorkspaceType="Event",
    NumEvents=100000,
    NumBanks=10,
    BankPixelWidth=15,
)
output_filename = f'~/Documents/nexus_processed_{mantid.__version__}.nxs'
start_time = time.time()
SaveNexusProcessed(InputWorkspace='new_ws', Filename=output_filename, CompressNexus=True)
save_time = time.time()
print(save_time - start_time)
```
This creates an even workspace of ~4.5GB in size.  Results as follows
version | workspace size | size on disk| time for save 
--|--|--|--
6.12.0 | 4500MB | 2.6GB | 93.3 s
6.13.1.1 | 4500MB | 2.6GB | 92.0 s
6.13.1.2 | 4500MB | 2.6GB | 92.s s

rboston628, 2025-09-30:
On main at current commit, I downgraded from 1.88 to 1.86:
``` bash
$ mamba list boost
> List of packages in environment: "/home/4rx/mambaforge/envs/mantid-developer"
>
>  Name                   Version  Build            Channel    
> ─────────────────────────────────────────
>   libboost               1.86.0   hed09d94_4       conda-forge
>   libboost-devel         1.86.0   hfcd1e18_4       conda-forge
>   libboost-headers       1.86.0   ha770c72_4       conda-forge
>   libboost-python        1.86.0   py311h1d5f577_4  conda-forge
>   libboost-python-devel  1.86.0   py311hf7b45f0_4  conda-forge
```
However, this did not fix the issue.

rboston628, 2025-09-30:
Trying `git bisect` again.

I can confirm the problem exists on this commit: 66c4338f441a3fe0f07303727b6154997a8e3297
I can confirm the problem does NOT exist on this commit: 282a6830cd13fcf95159e0b15f95f6aa0da9a1fc

rboston628, 2025-10-02:
The fix required undoing `chunkdims[0] = 1` that was introduced for data with an unlimited extent.

Here are results for good, bad, and fixed runs, using the above script with only 1.64527e+08

commit | ws size (MB) | save time (s) | load time (s) | memory ratio (disk : ws)
--|--|--|--|--
eea692d | 3765 | 3.041 | 3.898 | 1.00067
282a6830cd13fcf95159e0b15f95f6aa0da9a1fc | 3765 | 508.2 | 385.2 | 2.95032
w/ fix       | 3765 | 3.049 | 3.953 | 1.00067

The fix required changing a unit test.  There is a conflict in behavior with that unit test, and this defect was likely introduced to fit the test.

PR #40034 (2025-10-02): Fix `SaveMD` performance regression caused by unlimited data chunk dimensions
### Description of work

Fixes a performance regression in `SaveMD` that led to a many-fold increase in RAM usage, 3x increase in saved WS memory footprint, and a 30x increase in save and load times.

Closes #39954 

### To test:

Please run the test script in the original issue, and report save/load times.

---

### Reviewer

**Your comments will be used as part of the gatekeeper process.** Comment clearly on what you have checked and tested during your review. Provide an audit trail for any changes requested.

As p [... cut]
commit fdbc0079 (2025-10-02): Fix `SaveMD` performance regression caused by unlimited data chunk dimensions (#40034)

### Description of work

Fixes a performance regression in `SaveMD` that led to a many-fold
increase in RAM usage, 3x increase in saved WS memory footprint, and a
30x increase in save and load times.

Closes #39954 

### To test:

Please run the test script in the original issue, and report save/load
times.

---

### Reviewer

**Your comments will be used as part of the gatekeeper process.**
Comment clearly on what you have checked and tested during your review.
Provide an audit trail for any changes reques [... cut]