mantidproject/mantid #39985: Manual Testing Sliceviewer

MialLewis, 2025-09-25:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html

jhaigh0, 2025-09-29:
~ 90mins

warunawickramasingha, 2025-09-29:
The script for [Representation of integrated peaks](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#id10) gave this error
```
Error in execution of algorithm MaskBTP:
Instrument 'basic_rect' not in the allowed list
  at line 159 in 'C:/MantidNightlyInstall/plugins/python\algorithms/MaskBTP.py'
Can't execute MaskBTP algorithm for this instrument to set edge for IntegrateIfOnEdge option
```

MialLewis, 2025-10-03:
> The script for [Representation of integrated peaks](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#id10) gave this error
>
> ```
> Error in execution of algorithm MaskBTP: [...]

As per the test instructions, this error is expected.

closed 2025-10-03 by MialLewis