mantidproject/mantid #39992: Manual Testing M-slice

MialLewis, 2025-09-25:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Direct/MSliceTestGuide.html

MohamedAlmaki, 2025-09-29:
Minor docs updates:
in the default settings the tabs won't be enable unless you select the entry on the left
You need to minimise the plots instead of closing them in points one and two

MohamedAlmaki, 2025-09-29:
There are some resolution issues opening MSlice on my high resolution screen, and some UI elements appear so samll

[image]

[image]

RichardWaiteSTFC, 2025-09-29:
Interaction with ADS - rename one of the workspaces in the ADS and the name isn't changed in the MDHisto list in Mslice (but plotting still works). Delete the workspace in the ADS, workspace still there in MDHisto list in MSclice (plotting still works).

RichardWaiteSTFC, 2025-09-29:
Time ~1h30mins