mantidproject/mantid #39996: Manual Testing Error Reporter

MialLewis, 2025-09-25:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ErrorReporterTesting.html

RichardWaiteSTFC, 2025-09-29:
For me the first segfault I ran on IDAaaS error report window loaded (eventually - it took minutes) but quickly closed with following printed in the log
```
CheckMantidVersion-[Notice] CheckMantidVersion started
CheckMantidVersion-[Notice] CheckMantidVersion successful, Duration 0.17 seconds
errorreports (pystack analysis)-[Notice] Could not find recent enough ( < 30 seconds old) valid core dump file in {core_dumps_dir.as_posix()}
ICE default IO error handler doing an exit(), pid = 241258, errno = 32
```

RichardWaiteSTFC, 2025-09-29:
Test 2 - Didn't remember me from test 1(but worked for  @MohamedAlmaki)

RichardWaiteSTFC, 2025-09-29:
> Test 2 - Didn't remember me from test 1(but worked for [@MohamedAlmaki](https://github.com/MohamedAlmaki))

It did remember me in Test 3 (i.e. after I had submitted a report with `Yes, share information`)

RichardWaiteSTFC, 2025-09-29:
> > Test 2 - Didn't remember me from test 1(but worked for [@MohamedAlmaki](https://github.com/MohamedAlmaki))
> 
> It did remember me in Test 3 (i.e. after I had submitted a report with `Yes, share information`)

IN test 4 it remembered the public details form test 2 - i.e. it seems the details are only remembered if a report is submitted - is this expected?

RichardWaiteSTFC, 2025-09-29:
It would be helpful if it told us where this file is - there's a `Mantid.properties` file in the `bin` folder which did work for this test.
https://github.com/mantidproject/mantid/blob/d150656d2239278846b73d0ff5f5bdd4b75a9bc9/dev-docs/source/Testing/ErrorReporter-ProjectRecovery/ErrorReporterTesting.rst?plain=1#L122-L124

jhaigh0, 2025-10-01:
> [@jhaigh0](https://github.com/jhaigh0) - would you mind checking if [#39996 (comment)](https://github.com/mantidproject/mantid/issues/39996#issuecomment-3347265517) and [#39996 (comment)](https://github.com/mantidproject/mantid/issues/39996#issuecomment-3347270070) are expected/OK?

Just to confirm on here, not expected, we're looking into it, we think something has broken since this went in https://github.com/mantidproject/mantid/pull/39860