mantidproject/mantid #40004: Manual Test Mantid Plotting

MialLewis, 2025-09-25:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Make sure the Mantid Plotting is working correctly and each testing team follows below test instructions on separate test issues https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/plotting/index.html

jhaigh0, 2025-09-29:
Run this script
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
from matplotlib.colors import LogNorm

data = Load('MAR11060')

fig, axes = plt.subplots(subplot_kw={'projection':'mantid'})

# IMPORTANT to set origin to lower
c = axes.imshow(data, origin = 'lower', cmap='viridis', aspect='auto', norm=LogNorm())
cbar=fig.colorbar(c)
cbar.set_label('Counts ($\mu s$)$^{-1}$') #add text to colorbar

fig.show()
```

In the subsequent plot window, copy script to clipboard, I get this:

```
import matplotlib.pyplot as plt
from mantid.plots.utility import MantidAxType
import numpy as np
from matplotlib.colors import LogNorm
from matplotlib.ticker import LogLocator
from mantid.api import AnalysisDataService as ADS

data = ADS.retrieve('data')

fig, axes = plt.subplots(dpi=139.0, edgecolor='#ffffff', ncols=2, nrows=3, num='Figure 1', subplot_kw={'projection': 'mantid'})
cfill = axes[0][0].imshow(data, aspect='auto', cmap='viridis', distribution=False, label='_child0', origin='lower')
cfill.set_norm(LogNorm(vmin=0.35247836349331235, vmax=227.33333333333343))
# If no ticks appear on the color bar remove the subs argument inside the LogLocator below
cbar = fig.colorbar(cfill, ax=[axes[0][0]], ticks=LogLocator(subs=np.arange(1, 10)), pad=0.06)
cbar.set_label('Counts ($\\mu s$)$^{-1}$')
axes[0][0].tick_params(axis='x', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False})
axes[0][0].tick_params(axis='y', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False})
axes[0][0].set_xlabel('Time-of-flight ($\\mu s$)')
axes[0][0].set_ylabel('Spectrum')
axes[0][0].set_xlim([5.0, 20000.0])
axes[0][0].set_ylim([0.5, 922.5])

fig.show()
# Use plt.show() if running the script outside of Workbench
#plt.show()
# Scripting Plots in Mantid:
# https://docs.mantidproject.org/tutorials/python_in_mantid/plotting/02_scripting_plots.html
```

Produces the same plot but in a grid of six plots (`ncols=2, nrows=3`) with the other 5 empty.

MialLewis, 2025-09-29:
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
from mantid.plots._compatability import plotSpectrum #import needed outside Workbench

'''Processing'''

# Load the spectral range of the interest of the data
ws=Load(Filename="GEM40979.raw", SpectrumMin=431, SpectrumMax=750)

# Convert to dSpacing
ws=ConvertUnits(InputWorkspace=ws, Target= "dSpacing")

# Smooth the data
ws=SmoothData(InputWorkspace=ws, NPoints=20)

'''Plotting'''

# Plot three spectra
PLOT_WINDOW = plotSpectrum(ws, [0,1,2])

# Get Figure and Axes for formatting
fig, axes = plt.gcf(), plt.gca()

# Set the scales on the x- and y-axes
axes.set_xlim(4, 6)
axes.set_ylim(0, 5e3)

# Plot index 5
plotSpectrum(ws,5, window= PLOT_WINDOW, clearWindow= False)

# Alter the Y label
axes.set_ylabel('Neutron Counts ($\AA$)$^{-1}$')

# Optionally rename the curve titles
axes.legend(["bank2 sp-431", "bank2 sp-432", "bank2 sp-433", "bank2 sp-436"])

# Give the plot a title
plt.title("Oscillations", fontsize=20)
```

- Run above script
- Change y axis to symlog
- Plot is incorrect, until home is clicked.

jhaigh0, 2025-09-29:
```
data = Load('MUSR00015189.nxs')
``
plot `data_1` as a colourfill plot. Go to the settings and change the x axis to `log`. The plot doesn't the data correctly, if you press home it doesn't fix it. Changing the scale does fix the limits strangely.

This has been reproduced in 6.13.1

jhaigh0, 2025-10-01:
@MialLewis I can recreate it on Windows. It's not the scaling that's changed it's just the `Canvas Color`

[image]

MialLewis, 2025-10-02:
> [@MialLewis](https://github.com/MialLewis) I can recreate it on Windows. It's not the scaling that's changed it's just the `Canvas Color`
> 
> <img alt="Image" width="388" height="52" src="https://private-user-images.githubusercontent.com/35813666/496255841-23c2fcd9-4c1c-4a15-bad9-f6dba724d15c.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkzOTEwNTksIm5iZiI6MTc1OTM5MDc1OSwicGF0aCI6Ii8zNTgxMzY2Ni80OTYyNTU4NDEtMjNjMmZjZDktNGMxYy00YTE1LWJhZDktZjZkYmE3MjRkMTVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEwMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMDAyVDA3MzkxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkZDY2MTRmN2I5YmNiMmI5YmI0M2U0M2ViMjcyMjgyMWVjZTY1OGM5ZDA5Zjk3M2Y5OWFhMTFjNTAwYmMwODYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.-M74pStvq3I9_N2_ZY3beSLKfi0Ury5qg7CCEkwnXRY">

Ah, sorry I didn't read your post well!

closed 2025-10-03 by MialLewis