mantidproject/mantid #40011: Failure to load files on POLARIS

jclarkeSTFC, 2025-09-26:
**Original reporter:** Paul Henry

**Describe the bug**
From support email:
```
We have a very odd situation on Polaris at ISIS that we cannot read some of the current cycle data into Mantid.

We cannot read any data into Mantid after run 149511 – it was 149510 earlier today and we also had issues early in the cycle for calibration (earlier run numbers) that probably arises from the same issue.

Mantid throws an error (at end of email) that the data file does not exist even though we can see it in the data directory

The data directory is the ISIS data store, where the drive is correctly shared ([\\isis\inst$\NDXPOLARIS\Instrument](file:///inst$/NDXPOLARIS/Instrument)) on the two computers that we have replicated the error with

Further, if we map the specific cycle directory as a separate network drive and add that drive into Mantid user directory settings, it reads all data.

We are stumped.

Any help/suggestions appreciated.

Regards,

Paul
```
 ```
Error log below – using Mantid workbench 6.13

ValueError: Problem setting "Filename" in Load: Invalid value for property Filename (list of str lists) from string "POLARIS149512": Unable to find file matching the string "POLARIS00149512", please check the data search directories. Could not find data cache directory: /data/instrument

No file found.

If the file has been found but you got this error, you might not have read permissions or the file might be corrupted.

If the file has not been found, you might have forgotten to add its location in the data search directories.

File "P:/MantidAnalysisOnPDrive/user_focus_25_3_PFH.py", line 21, in <module>

polaris_object.focus(mode="PDF",do_absorb_corrections=False,do_van_normalisation=True,

File "C:\MantidInstall\scripts\Diffraction\isis_powder\polaris.py", line 38, in focus

return self._focus(

^^^^^^^^^^^^

File "C:\MantidInstall\scripts\Diffraction\isis_powder\abstract_inst.py", line 93, in _focus

focused_runs = focus.focus(

^^^^^^^^^^^^

File "C:\MantidInstall\scripts\Diffraction\isis_powder\routines\focus.py", line 30, in focus

return _individual_run_focusing(

^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\MantidInstall\scripts\Diffraction\isis_powder\routines\focus.py", line 411, in _individual_run_focusing

ws = common.load_current_normalised_ws_list(run_number_string=run, instrument=instrument)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\MantidInstall\scripts\Diffraction\isis_powder\routines\common.py", line 359, in load_current_normalised_ws_list

raw_ws_list = load_raw_files(run_number_string=run_number_string, instrument=instrument, file_ext=file_ext)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\MantidInstall\scripts\Diffraction\isis_powder\routines\common.py", line 602, in load_raw_files

load_raw_ws = _load_list_of_files(file_name_list)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\MantidInstall\scripts\Diffraction\isis_powder\routines\common.py", line 676, in _load_list_of_files

loaded_ws = mantid.Load(Filename=file_name, OutputWorkspace=file_name)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\MantidInstall\bin\Lib\site-packages\mantid\simpleapi.py", line 157, in Load
```

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version 6.13

**Additional context**

jclarkeSTFC, 2025-09-26:
Logs from Paul:

```
Logging set to debug priority

Looking for POLARIS at ISIS.

Instrument 'POLARIS' found as POLARIS at ISIS.

1 Finished algorithms removed from the managed algorithms list. 1 remaining.

Looking for POLARIS at ISIS.

Instrument 'POLARIS' found as POLARIS at ISIS.

Looking for POLARIS at ISIS.

Instrument 'POLARIS' found as POLARIS at ISIS.

vector findRun('POLARIS00149512', exts[18])

Looking for POLARIS at ISIS.

Instrument 'POLARIS' found as POLARIS at ISIS.

getExtension(POLARIS00149512, exts[7])

Failed to find extension. Just using last '.'

toInstrumentAndNumber(POLARIS00149512)

Looking for POLARIS at ISIS.

Instrument 'POLARIS' found as POLARIS at ISIS.

get archive search for the facility...ISISDataSearch

getFullPath(POLARIS00149512.nxs)

Searching for POLARIS00149512.nxs in V:/

Searching for POLARIS00149512.nxs in C:\Users\zlx12256\AppData\Roaming\mantidproject\mantid\instrument\

Searching for POLARIS00149512.nxs in C:/MantidInstall/instrument/

getFullPath(POLARIS00149512.raw)

Searching for POLARIS00149512.raw in V:/

Searching for POLARIS00149512.raw in C:\Users\zlx12256\AppData\Roaming\mantidproject\mantid\instrument\

Searching for POLARIS00149512.raw in C:/MantidInstall/instrument/

getFullPath(POLARIS00149512.sav)

Searching for POLARIS00149512.sav in V:/

Searching for POLARIS00149512.sav in C:\Users\zlx12256\AppData\Roaming\mantidproject\mantid\instrument\

Searching for POLARIS00149512.sav in C:/MantidInstall/instrument/

getFullPath(POLARIS00149512.add)

Searching for POLARIS00149512.add in V:/

Searching for POLARIS00149512.add in C:\Users\zlx12256\AppData\Roaming\mantidproject\mantid\instrument\

Searching for POLARIS00149512.add in C:/MantidInstall/instrument/

getFullPath(POLARIS00149512.nxspe)
[... 576 lines cut ...]

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\MantidInstall\scripts\Diffraction\isis_powder\routines\common.py", line 359, in load_current_normalised_ws_list

raw_ws_list = load_raw_files(run_number_string=run_number_string, instrument=instrument, file_ext=file_ext)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\MantidInstall\scripts\Diffraction\isis_powder\routines\common.py", line 602, in load_raw_files

load_raw_ws = _load_list_of_files(file_name_list)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\MantidInstall\scripts\Diffraction\isis_powder\routines\common.py", line 676, in _load_list_of_files

loaded_ws = mantid.Load(Filename=file_name, OutputWorkspace=file_name)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\MantidInstall\bin\Lib\site-packages\mantid\simpleapi.py", line 157, in Load

raise ValueError(
```

Anders-Markvardsen, 2025-10-22:
Notes from isis automation steering group meeting earlier today:
* About 140 MARI automation jobs related to failure to load from the archive
* All Vesuvio runs failed, which was speculated might also relate to loading from archive
* It was noted that no issue when loading data from DAaaS disk space
* It was also mentioned that Vesuvio had issues with Journal Viewer - it was speculated that this journal viewer issue may also be related to issue with archive

cailafinn, 2025-10-29:
Thanks for looking into that Freddie, and for the info about why! I can confirm that Mantid is now finding the paths to the files correctly :)